FPGAwars/icestudio #310: JSHint errors when running npm run dist

BryceBeagle, 2019-02-10:
When I run `npm run dist` I get errors in the jshint grunt task and the build fails:

```
┌─[ignormies][~/git/icestudio]
└──▪ npm run dist  

> icestudio@0.4.0-dev dist /home/ignormies/git/icestudio
> grunt dist

Running "checksettings" task

Running "jshint:all" (jshint) task

   app/scripts/controllers/menu.js
     71 |      let val = queryStr.replace(regex, "$1");
                                                     ^ Strings must use singlequote.
     72 |      let params = (val == queryStr) ? false : val;
                                   ^ Expected '===' and instead saw '=='.
     77 |          if(typeof gui.App.argv ==='undefined') gui.App.argv=[];
                                                          ^ Expected '{' and instead saw 'gui'.
     69 |      let queryStr = unescape(window.location.search) + '&';
                              ^ 'unescape' is not defined.
   app/scripts/services/graph.js
     62 |        let win_width = window.get().width;
                              ^ Identifier 'win_width' is not in camel case.
     63 |        let win_height = window.get().height;
                               ^ Identifier 'win_height' is not in camel case.
     68 |          width: win_width - 2 * margin,
                                   ^ Identifier 'win_width' is not in camel case.
     69 |          height: win_height - menuFooterHeight - 2 * margin
                                     ^ Identifier 'win_height' is not in camel case.
    104 |        $('.paper.joint-theme-default>svg').attr('height',win_height);
                                                                             ^ Identifier 'win_height' is not in camel case.
    105 |        $('.paper.joint-theme-default>svg').attr('width',win_width);
                                                                           ^ Identifier 'win_width' is not in camel case.
   app/scripts/services/tools.js
    475 |                var msg = '';
                             ^ 'msg' is already defined.
    476 |                var line = parseInt(matchError[1]);
                             ^ 'line' is already defined.
    477 |                var type = matchError[3].toLowerCase();
                             ^ 'type' is already defined.
    478 |                var preContent = matchError[4];
                             ^ 'preContent' is already defined.
   app/scripts/services/utils.js
    884 |            if(params===false) params={};
                                        ^ Expected '{' and instead saw 'params'.
    901 |        var new_win = gui.Window.open(url, {
                            ^ Identifier 'new_win' is not in camel case.
    902 |            "new-instance" : true,
                                   ^ Strings must use singlequote.
    903 |            "new_instance" : true,
                                   ^ Strings must use singlequote.
    904 |            "position": "center",
                               ^ Strings must use singlequote.
    904 |            "position": "center",
                                         ^ Strings must use singlequote.
    905 |            "toolbar": false,
                              ^ Strings must use singlequote.
    906 |            "width": 900,
                            ^ Strings must use singlequote.
    907 |            "height": 600,
                             ^ Strings must use singlequote.
    908 |            "show": true,
                           ^ Strings must use singlequote.
    901 |        var new_win = gui.Window.open(url, {
                     ^ 'new_win' is defined but never used.

>> 25 errors in 29 files
Warning: Task "jshint:all" failed. Use --force to continue.

Aborted due to warnings.
npm ERR! code ELIFECYCLE
npm ERR! errno 3
npm ERR! icestudio@0.4.0-dev dist: `grunt dist`
npm ERR! Exit status 3
npm ERR! 
npm ERR! Failed at the icestudio@0.4.0-dev dist script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

umarcor, 2019-12-25:
@BryceBeagle, I think that those errors are not reproducible with branch `develop` (which is good). However, you might get other linting errors in `app/scripts/services/tools.js`.

This should probably be checked through a CI service, but unfortunately there seems to be no working setup for this repo. There have been two runs (only) in the last year: https://travis-ci.org/FPGAwars/icestudio/builds

cavearr, 2020-01-29:
on 0.5.1 dev branch this jshint issues are solved.

closed 2020-01-29 by cavearr