FPGAwars/icestudio #314: unknown module type SB_SPRAM256KA

janrinze, 2019-03-07:
Apparently the tools with the latest releases are outdated.
Since the app requires an old version of node.js (included in the build) it's not straight forward how to get up to date tools.

Any hints on how to solve this? (Would love to have icestudio to use the manually built tools in /usr/local/bin using the build instructions on project icestorm)

janrinze, 2019-04-07:
a workaround:

~/.icestudio/apio/packages/toolchain-icestorm/bin/ is where the binary tools reside.

Building yosys, arachnepnr, icestorm and nextpnr from source with the prefix ~/.icestudio/apio/packages/toolchain-icestorm and doing make install can update them to the most recent version.

cavearr, 2020-01-29:
Hi! we are updating icestudio , 0.5.1 dev branch has the new toolchain, but maximun node version tu build icestudio is 10.17.x

We are working on a new icestudio version with upstream tool versions, but we are more slowly that we want.

closed 2020-01-29 by cavearr