FPGAwars/icestudio #318: Error libusb_open en debian 9 tarjeta Alhambra II

avlixa, 2019-04-19:
Hola,
He instalado icestudio en debian 9 y al intentar cargar un código de prueba me da el error
ftdi_usb_get_strings failed: - 4 ( libusb_open() failed)

He visto este error documentado para windows 10 pero no encuentro nada para debian.  En windows 10 la placa me funciona bien.

También me he dado cuenta que al actualizar la toolchain siempre me dice que no coincide, desconozco si tiene algo que ver.

Saludos.
[image]
[image]

JuanMiguelTorre, 2020-02-21:
Hola, a mi me ocurre lo mismo lo único que con la iCEstick Evaluation Kit.

myzinsky, 2021-01-03:
same problem on macOS

LBertrandDC, 2021-08-14:
Hello, I just noticed the same on Windows 10 Enterprise (what my college installs on staff computers). 
Using 0.7.1w202108130708

Running the lsftdi.exe binary directly from PowerShell:
>  PS C:\Users\Louis\\.icestudio\apio\packages\tools-oss-cad-suite\bin> .\lsftdi.exe
>  Number of FTDI devices found: 1
>  Checking device: 0
>  ftdi_usb_get_strings failed: -4 (libusb_open() failed)

The problem might caused by pilot error. When the Zadig driver utility appears, there are several choices of USB devices and the user (me!) did not select "Lattice FTUSB Interface Cable (Interface 0)". I can't remember because I wasn't paying attention to that detail.
I ran
> apio drivers --ftdi-enable

from the venv under .icestudio in the home directory/folder and that seems to have corrected the problem.
Also, it helps if you unplug as many USB devices as possible to shorten the list of devices.

Obijuan, 2022-05-02:
Could you please try the lates Icestudio wip? You can download it from here:

https://downloads.icestudio.io/

I am cleaning old issues. I will close this issue. Please, feel free to open a new one with updated information if the error remains
Thanks for the feedback

closed 2022-05-02 by Obijuan

LBertrandDC, 2022-05-02:
Hi @Obijuan, I can program the Lattice iCEstick, no problems, using either WIP snapshot from today 0.9.2w20220502{0505|0905}
Also, random fact if that helps, I tried (from the Zadig utility) the WinUSB driver and it also worked.

Obijuan, 2022-05-02:
Great! Thanks for the feedback 🙂️