FPGAwars/icestudio #320: Is any ECP5 board in plan?

goran-mahovlic, 2019-05-03:
We would love to have support for ULX3S
https://hackaday.io/project/159108-ulx3s-powerful-ecp5-board-for-open-source-fpga

drtrigon, 2019-06-10:
That would give us 2nd board with icestudio & Arduino IDE support, beside the Alhambra II (and similar): https://github.com/FPGAwars/icestudio/issues/321 and https://github.com/drtrigon/fpgarduino-icestorm

janrinze, 2019-09-25:
prjtrellis is supported by apio.
So if the constraints file can be generated and the right version of nextpnr can be included, I don't see what could be the big hurdle here.

Obijuan, 2019-12-23:
@mmicko has already added suport for ulx3s boards in the devel branch. We are working on it now

cavearr, 2020-01-29:
At v0.5.1 dev branch we have support for ecp5 boards

closed 2020-01-29 by cavearr