FPGAwars/icestudio #321: Firmware update/upload via Arduino IDE (e.g. for picorv32)

drtrigon, 2019-05-07:
Thanks to https://github.com/cliffordwolf/picorv32/issues/92 we have a picorv32 version for icestudio. Now I would like to be able to use the Arduino IDE in order to upload firmware code to the processor (instead of using a Makefile based toolchain). In fact this was already mentioned in https://github.com/cliffordwolf/picorv32/issues/92; "The firmware should be uploaded in the flash memory using iceprog, outside of icestudio. In future versión we will integrate it into icestudio as well".

Furthermore it would also allow to use the Arduino IDE syntax/api and therefore the examples, libraries, etc. already available. Basically a lot of existing code could be adopted and used.

This would be very similar to the FPGArduino project (see http://www.nxlab.fer.hr/fpgarduino/) with the subtile difference that it will not use a bootloader but incorporate iceprog directly. (For details on that work see https://github.com/f32c/arduino/issues/32 - caution this thread is really long!)

The current status of this work can be seen in https://github.com/drtrigon/fpgarduino-icestorm (the naming is not appropriate anymore). Currently I am working on the digitalRead and Serial (port) examples.

Thus I would like to propose this feature to be integrated into the https://github.com/FPGAwars/icestudio/wiki/Wishlist:-proposed-features (outdated?) respective for the v0.4.0 release. What do you think? Feedback on this would be very welcome.

Further References:
* https://github.com/cliffordwolf/picorv32/issues/95

drtrigon, 2019-06-06:
Latest commit: https://github.com/drtrigon/fpgarduino-icestorm/commit/09d6520db16b8c602c879129a4065a3a33f698aa

What does work now:
* Verify (building)
* Upload
* Serial Monitor using any baudrate, default is 115200 (needs setting the correct port)
* Arduino IDE examples:
  * 01.Basics
    * BareMinimum: compiles
    * Blink: pinMode dummy for leds, digitalWrite on leds also using full 8-bit uint8_t, delay
    * DigitalReadSerial: Serial.println, Serial.begin all baudrates, pinMode (0-7), digitalRead (0-15)
    * AnalogReadSerial: analogRead (A0, 4Hz, bits increased from 8 to 10 by multiplying with 4)
    * ReadAnalogVoltage: (voltage calculation wrong 5.0->3.3)
    * (TODO: Fade - needs PWM)
  * 02.Digital
    * BlinkWithoutDelay: millis, micros, (delay refactored to use micros)
    * Button: (works after changing ledPin 13->LED_BUILTIN)
    * Debounce: (works after changing ledPin 13->LED_BUILTIN)
    * StateChangeDetection: (works after changing ledPin 13->LED_BUILTIN)
  * 03.Analog
    * AnalogInput: (works after changing ledPin 13->LED_BUILTIN)

Obijuan, 2019-06-07:
That's awesome! Thanks for sharing! I will try to test it as soon as possible. Keep up the good work! :-)

drtrigon, 2019-06-09:
The most recent commit now supports a lot more examples as it includes analogWrite (PWM), Serial.read and digitalWrite for all pins now: https://github.com/drtrigon/fpgarduino-icestorm/commit/7a31a42964c6704b8ea3ffce2b8730932badf305

Tried to get configurable pull-ups on the GPIO pins, but I only get a `PULLUP with non-constant value`.

drtrigon, 2019-08-07:
Btw.: What's the state with Lattuino (https://github.com/FPGALibre/fpgacores) regarding icestudio support?