FPGAwars/icestudio #322: Inconveniente con icestudio en Ubuntu18.0.4

ebl802, 2019-05-13:
Hola 
Estoy tratando de instalar Icestudio en una instalación nueva de Ubuntu18.0.4, pero al intentar ejecutar "icestudio-0.4.0-dev-linux64.AppImage" recibo el siguiente mensaje:

installed: X-AppImage-BuildId=cd8ad3cf-8c58-4d3f-9ab6-af0a92307596 image: X-AppImage-BuildId=cd8ad3cf-8c58-4d3f-9ab6-af0a92307596
/tmp/.mount_MRElIT/usr/bin/icestudio: error while loading shared libraries: libgconf-2.so.4: cannot open shared object file: No such file or directory

Aguien me puede dar una mano.
Saludos

Obijuan, 2019-05-13:
Aquí está la solución :

https://github.com/Obijuan/digital-electronics-with-open-FPGAs-tutorial/wiki/V%C3%ADdeo-2:-%C2%A1Poniendo-en-marcha-Icestudio!#nota-sobre-ubuntu-1804

Saludos, Obijuan

closed 2019-05-21 by Obijuan

forensicgarlic, 2019-08-17:
I ran into this same issue using Ubuntu 19.04, and the same fix worked.

a2800276, 2019-08-29:
The workaround on Ubuntu is, well, to install the missing gconf lib. On Ubuntu:

     apt install libgconf-2-4

I'm confused that the issue is thereby resolved? Should an AppImage contain all it's dependencies? It would certainly make it easier to get started if it did.

Obijuan, 2019-08-29:
> I'm confused that the issue is thereby resolved? Should an AppImage contain all it's dependencies? It would certainly make it easier to get started if it did.

Yes, you are right. It is bug that has to be solved

jpfersich, 2019-08-30:
The only issue is how many different Linux distros are you going to support? Ubuntu 18.04 is a very broken distribution , installing software and libraries that formerly came with Ubuntu 16.04 is a constant hassle. I don’t know why Canonical chose to cripple Ubuntu 18.04. It’s gotten to the point that I’m going to install Fedora on a machine to see if it’s less of a hassle to maintain. Ubuntu 18.04 sucks.

/————————————————————/
For encrypted mail use jgpfersich@protonmail.com
Get a free account at ProtonMail.com
Web: www.objectnets.net and www.objectnets.org

a2800276, 2019-08-30:
@jpfersich I agree wholeheartedly that it's not necessary to support everything any user decides to throw at you. Informally though, it seems that if a vendor chooses a _single_ distribution to support, they usually go with Ubuntu LTS, as that seems to have the largest current marketshare. (I may be biased on both counts...)

Also, I'm not very well versed concerning the specifics of AppImage, but I thought that it was supposed to represent something like a static build of an application plus all support files bundled into an archive. My impression was that it's supposed to contain _all_ dependencies and shouldn't make assumptions on what may already be installed on a particular users machine.

I may very well be mistaken, and obviously that would be a bit wasteful, but a lot of vendors seem to be going down the everything-and-the-kitchen-sink approach these days, either using static linking like golang or shipping docker containers...