FPGAwars/icestudio #324: Exportar a Verilog: Bloque basic.memory no exporta su contenido

jorgef1962, 2019-06-11:
Al exportar a verilog un bloque de memoria, se crea un parámetro con tipo "archivo" pero el archivo con los datos no és creado.

En los archivos adjuntos:
El bloque "Cadena" genera una línea de verilog [parámetro v801bc3 = "v801bc3.list"] pero no se crea el archivo "v801bc3.list".

Los archivos adjuntos son de Icestudio 0.3.3 en Ubuntu 18.04.2 LTS (64 bits)
Pasa lo mismo con Icestudio 0.3.0 en Windows 10 home edition (64 bits)

El proyecto es uno ejemplo de los tutoriales de la collection "Academia-Jedi-HW-30" de los tutoriales de "obijuan" (Tutorial 30 ejemplo 6.2).

Saludos y muchas gracias por una herramienta de alta calidad
Jorge Ferreira
[06-2-Tx-string-bloque.zip](https://github.com/FPGAwars/icestudio/files/3277475/06-2-Tx-string-bloque.zip)

Obijuan, 2019-06-18:
He también he conseguido reproducir el problema. Es un bug

Obijuan, 2019-06-18:
Garcias por el informe @jorgef1962 :-)

cavearr, 2025-01-19:
Cierro este bug, ahora al hacer build se generan los verilog y los .list correspondientes. La última wip en https://downloads.icestudio.io

closed 2025-01-19 by cavearr