FPGAwars/icestudio #328: Verilog submodule inside code block

henrydang80, 2019-08-14:
Hello,

Thank you very much for making a very interesting FPGA development tool.

Since the tool has a limited component to build a big FPGA project, I would like to use "Code" block to make some custom components. However, it seems to support only one module behavior implementation (by checking generated main.v) and I don't know how to add submodule into the Code block and get compilation success.

Please show me is there any way to use Code block with multiple verilog submodule inside it ?

janrinze, 2019-09-25:
To have a verilog module included add on the top of the code block:

// @include mymodule.v

That will ensure that the module is included.
The file mymodule.v needs to be in the same directory

kamejoko80, 2019-09-25:
Hi Janrinze,

I’ve tried following your idea and it works.
Thank you very much.

cavearr, 2020-01-29:
Hi @henrydang80 with the new versions we add the feature of create and editing new submodules. Now we have multilevel edition!

closed 2020-01-29 by cavearr