FPGAwars/icestudio #331: [Icestudio 0.5-inception][Bug] Verilog error notification icon is not clear when the errors are fixed

Obijuan, 2019-08-28:
This is  a minor bug. I will describe it in Spanish

Dentro de un submódulo al darle a Verificar aparece un icono en la parte superior derecha de la ventana, si hay algún error

Pero una vez que se ha eliminado el error y se vuelve a dar a verifcar, el icono no desaparece

Es un bug menor. Si sales del módulo y vuelves a entrar, desaparece

[image]

cavearr, 2019-08-28:
Fixed in my devel repository branch "submodules"

https://github.com/cavearr/icestudio

closed 2019-08-28 by cavearr

Obijuan, 2019-08-28:
Confirmed! The bug is gone!
Thanks a lot!   :-)