FPGAwars/icestudio #334: VCLK from submodule not propagated to main module

crohrandt, 2019-08-30:
I'm currently preparing a lecture on digital circuits for university classes. Therefore I want to have the basic flipflop structures as submodules. It is working quite well for latches, but then I want to introduce a rising edge detector as a submodule using some verilog flip-flops triggered by the vclk (board attached clock source).

[image]

If I directly program the PED on the TinyFPGA Bx board I can verify the function on an oscilloscope. If I than use this module in my own flip-flop module, there is no fpga flip-flip synthesized.

[image]
In the command output I get: 

    yosys -p "synth_ice40 -blif hardware.blif" -q main.v
    Warning: Wire main.\v7dabc5.vec866b.v526aa2.clk is used but has no driver.
    Warning: Wire main.\v7dabc5.v6b7f60.v526aa2.clk is used but has no driver.
    Warning: Wire main.\v7dabc5.v10b538.v526aa2.clk is used but has no driver.
    Warning: Wire main.\v7dabc5.v022335.v526aa2.clk is used but has no driver.

After having a look in the exported verilog code I can see that the embedded vclk wire is not propagated to the main module. So the fpga flip-flops get optimized out as they don't have a connected clk input.

[FFD.txt](https://github.com/FPGAwars/icestudio/files/3559586/FFD.txt)
[PED.txt](https://github.com/FPGAwars/icestudio/files/3559589/PED.txt)

Obijuan, 2019-08-30:
When creating a module, you should explicitly provide an input pin for the clock.

Here you can see an example. We are creating a module with two D-flip-flops:

[image]

Notice the upper pin: it is the clock, and you should connect it to the clk input of the DFFs

For creating a clk input pin just check "show clock":

[image]

When you add your module, you will see , the new clk input. Icestudio wil connect it automatically to the system clock (but it is only done in the top entity)

[image]

Try it with your design. Please, write in this issue the results or further problems. I will keep it open until it works for you :-)

crohrandt, 2019-08-30:
Thanks for your quick response. I just tried your suggestion and it works very well. 
Although it is a little bit confusing to have 2 clock inputs, but that's okay for me.

[image]

I couldn't think of another solution to insert some kind of delay for the edge detector than to put some flip-flops in one path. First I tried a sequence of several inverters, but they were optimized out. Do you know of a possibility to force the synthesizer to insert several LUTs although they seem to be useless?

closed 2019-08-30 by crohrandt