FPGAwars/icestudio #338: Tablas en modo "Local parameter".

Democrito, 2019-09-15:
Hola!

Me he encontrado con algo muy extraño, no sé si es que yo estoy haciendo algo mal.

Te adjunto dos circuitos casi iguales, sólo se diferencian en que uno va con tablas en el mismo circuito y en otro con "Local parameter"; este segundo no funciona, da datos repetitivos.

El circuito es un simple conversor binario a ASCII a través del serial.
Saludos!

[Tablas_y_local_parameter.zip](https://github.com/FPGAwars/icestudio/files/3614072/Tablas_y_local_parameter.zip)
.

Obijuan, 2019-09-15:
¡Hola Demócrito!

He probado tus ejemplos y efectivamente el segundo no funciona. 

Sin embargo, he partido del que sí funciona, he copiado las tablas de los parámetros sobre las que están dentro de los bloques y las he convertido en locales y funciona bien. 

Te adjunto el fichero para que lo pruebes

[Version-ok-local.zip](https://github.com/FPGAwars/icestudio/files/3614183/Version-ok-local.zip)

Los parámetros locales con memorias están funcionando. Parece que el problema está en cómo se ha creado el que no funciona. Eso es lo que tenemos que descubrir para acorralar el bug. Si consigures reproducir cómo llegaste al que no funciona a partir del que sí sería genial, así podemos encontrar qué paso es el que no va bien

Obijuan, 2019-09-15:
Por cierto, ¡Me ha encantado el ejemplo!! :-)  Es super sencillo y didáctico!

Democrito, 2019-09-15:
Hola Juan!

Hice muchas pruebas antes de enviar esta "Issue" y no conseguí ver la razón de ello. Y ahora he tomado el que no funciona, les eliminé las tablas de memoria y les volví a poner sus datos, y ha funcionado bien.

Otra prueba: tenía esos bloques aparte, los he vuelto a colocar y sigue funcionando bien... No lo entiendo, pero si me vuelve a suceder caminaré sobre mis pasos a ver qué puede ser...

Gracias y saludos!

Obijuan, 2019-09-16:
Estoy seguro de que hay un bug ahí escondido porque yo algunas veces he
notado efectos similares, pero al intentar replicarlo todo iba bien

La clave está en lograr aislarlo  y saber el patrón de acciones que lo
provocan

Seguiremos buscando

Muchas gracias por las pruebas y por reportar los problemas. ¡Es súper
necesario! 😉

cavearr, 2025-01-09:
Cierro esta issue, si es necesario volver a abrir.

closed 2025-01-09 by cavearr