FPGAwars/icestudio #339: Please document how to use gtkwave simulation with icestudio

drtrigon, 2019-09-19:
I would like to use icestudio in class e.g. to demonstrate the simulation of logic gates. Ultimatively I would like to create logic tables with the students.

I found following two descriptions but - unfortunately - do not speak spanish:
* https://groups.google.com/forum/#!topic/fpga-wars-explorando-el-lado-libre/E-EwTaCl01Q
* http://haudahau.com/vadedos/?p=1497 also after using google translator I am still confused. E.g. using the gtkwave export option alone does not work, as I understand I have to use `apio sim`. Installing apio according to https://apiodoc.readthedocs.io/en/stable/source/installation.html did not work under kubuntu 18.04 as there was no `apio` command available in terminal afterwards.

As mentioned I would be very eager to see documentation in English in order to understand what steps I need to do. (Also cool would be a simple menu entry to simulate and open gtkwave, but we can leave this for the next issue to open ;)

drtrigon, 2019-10-02:
Ok, I figured it out, I had an issue with the apio install path (due to python3 vs. python2 stuff).

https://translate.google.com/translate?hl=en&sl=es&u=http://haudahau.com/vadedos/%3Fp%3D1497 is quite useful.

csylvain, 2020-01-23:
running "apio sim" got me on the right path. needed to "apio install" some things. then needed to know i should look at the output of "apio boards -l". 

after that, learned something important and the reason i am writing this note:
(for the 0.5.1-nightly windows release of icestorm)
spaces (' ' chars) in project file names (like "B A Z.ice") is BAD.

you will end up with
```
fatal error: too many command-line arguments
```
when doing "apio build" because arachne-pnr dies. the parameters given to it are not quoted nor are the escaped.

zeroeth, 2020-02-03:
I'm wondering if this simulation workflow could be internal to icestudio somehow? Figuring out the export, edit testbench, run, view steps took some guesswork!

Maybe even embed something like Pulseview and provide an stimulus input panel/timeline!