FPGAwars/icestudio #348: Support for the Alchitry Cu?

kallisti5, 2019-11-01:
https://alchitry.com/collections/all/products/alchitry-cu-fpga-development-board

* Open Source Design
* iCE40-HX8K
* 8 LED's
* One push button
* 79 IO pins

kallisti5, 2019-11-01:
I started mapping out the pins, does this all look correct?

[pinout.pcf.txt](https://github.com/FPGAwars/icestudio/files/3799468/pinout.pcf.txt)

Obijuan, 2019-11-05:
For supporting a new board in icestudio you need to include it first in Apio
https://github.com/FPGAwars/apio

kallisti5, 2019-11-07:
apio already has support:

https://github.com/FPGAwars/apio/commit/690a614a4734699091cbf3811382e5efbda8e610

nyyirs, 2021-06-11:
hello guys, any news on this? I have an Alchitry Cu as well and I can contribute. Is there a doc on how to create a new board?

John-Westlake, 2021-11-20:
Found a case sensitive error in the pin define for C9 in file "pinout.json" :-

{"type":"inout","name":"B8","value":"c9"},

Should be (capital "C") :-

{"type":"inout","name":"B8","value":"C9"},

Cannot upload json file type here, so here's the corrected and reordered pinout.json for the Alchitry CU

[{"type":"input","name":"CLK","value":"P7"},
{"type":"inout","name":"A2","value":"M1"},
{"type":"inout","name":"A3","value":"L1"},
{"type":"inout","name":"A5","value":"J1"},
{"type":"inout","name":"A6","value":"J3"},
{"type":"inout","name":"A8","value":"G1"},
{"type":"inout","name":"A9","value":"G3"},
{"type":"inout","name":"A11","value":"E1"},
{"type":"inout","name":"A12","value":"D1"},
{"type":"inout","name":"A14","value":"C1"},
{"type":"inout","name":"A15","value":"B1"},
{"type":"inout","name":"A17","value":"D3"},
{"type":"inout","name":"A18","value":"C3"},
{"type":"inout","name":"A20","value":"A1"},
{"type":"inout","name":"A21","value":"A2"},
{"type":"inout","name":"A23","value":"A3"},
{"type":"inout","name":"A24","value":"A4"},
{"type":"inout","name":"A27","value":"A5"},
{"type":"inout","name":"A28","value":"C5"},
{"type":"inout","name":"A30","value":"D5"},
{"type":"inout","name":"A31","value":"C4"},
{"type":"inout","name":"A33","value":"D4"},
{"type":"inout","name":"A34","value":"E4"},
{"type":"inout","name":"A36","value":"F4"},
{"type":"inout","name":"A37","value":"F3"},
{"type":"inout","name":"A39","value":"H4"},
{"type":"inout","name":"A40","value":"G4"},
{"type":"inout","name":"A42","value":"H1"},
{"type":"inout","name":"A43","value":"H3"},
{"type":"inout","name":"A45","value":"K3"},
{"type":"inout","name":"A46","value":"K4"},
{"type":"inout","name":"A48","value":"N1"},
{"type":"inout","name":"A49","value":"P1"},
{"type":"inout","name":"B2","value":"A6"},
{"type":"inout","name":"B3","value":"A7"},
{"type":"inout","name":"B5","value":"A10"},
{"type":"inout","name":"B6","value":"A11"},
{"type":"inout","name":"B8","value":"C9"},
{"type":"inout","name":"B9","value":"C10"},
{"type":"inout","name":"B11","value":"A12"},
{"type":"inout","name":"B12","value":"B14"},
{"type":"inout","name":"B15","value":"D14"},
{"type":"inout","name":"B17","value":"E14"},
{"type":"inout","name":"B18","value":"E12"},
{"type":"inout","name":"B20","value":"F14"},
{"type":"inout","name":"B21","value":"G14"},
{"type":"inout","name":"B23","value":"H12"},
{"type":"inout","name":"B24","value":"J12"},
{"type":"inout","name":"B27","value":"H11"},
{"type":"inout","name":"B28","value":"G11"},
{"type":"inout","name":"B30","value":"G12"},
{"type":"inout","name":"B31","value":"F12"},
{"type":"inout","name":"B33","value":"F11"},
{"type":"inout","name":"B34","value":"E11"},
{"type":"inout","name":"B36","value":"D12"},
{"type":"inout","name":"B37","value":"D11"},
{"type":"inout","name":"B39","value":"C12"},
{"type":"inout","name":"B40","value":"C11"},
{"type":"inout","name":"B42","value":"D10"},
{"type":"inout","name":"B43","value":"D9"},
{"type":"inout","name":"B45","value":"D7"},
{"type":"inout","name":"B46","value":"D6"},
{"type":"inout","name":"B48","value":"C7"},
{"type":"inout","name":"B49","value":"C6"},
{"type":"inout","name":"C2","value":"M3"},
{"type":"inout","name":"C3","value":"M4"},
{"type":"inout","name":"C5","value":"L4"},
{"type":"inout","name":"C6","value":"L5"},
{"type":"inout","name":"C8","value":"M6"},
{"type":"inout","name":"C9","value":"M7"},
{"type":"inout","name":"C11","value":"L9"},
{"type":"inout","name":"C39","value":"P10"},
{"type":"inout","name":"C40","value":"P9"},
{"type":"inout","name":"C42","value":"L8"},
{"type":"inout","name":"C43","value":"L6"},
{"type":"inout","name":"C45","value":"P5"},
{"type":"inout","name":"C46","value":"P4"},
{"type":"inout","name":"C48","value":"P3"},
{"type":"inout","name":"C49","value":"P2"},
{"type":"output","name":"LED0","value":"J11"},
{"type":"output","name":"LED1","value":"K11"},
{"type":"output","name":"LED2","value":"K12"},
{"type":"output","name":"LED3","value":"K14"},
{"type":"output","name":"LED4","value":"L12"},
{"type":"output","name":"LED5","value":"L14"},
{"type":"output","name":"LED6","value":"M12"},
{"type":"output","name":"LED7","value":"N14"},
{"type":"output","name":"TX","value":"P14"},
{"type":"input","name":"RX","value":"M9"},{"type":"output","name":"SS","value":"P13"},{"type":"output","name":"SCK","value":"P12"},{"type":"output","name":"MOSI","value":"M11"},{"type":"input","name":"MISO","value":"P11"},{"type":"input","name":"CRESET","value":"L10"},{"type":"output","name":"CDONE","value":"M10"},
{"type":"inout","name":"RESET","value":"P8"}]

Obijuan, 2022-05-02:
The Alchitry Cu board is already supported. I will close this issue
Thanks for the feedback

closed 2022-05-02 by Obijuan