FPGAwars/icestudio #355: Add subblocks as references

olofk, 2019-11-21:
When I add a block to a project, icestudio seems to make an internal copy of that block, so when I update the original block, it is not changed in the new project. This makes it hard to reuse components. It would be very useful to add the block as a reference to another .ice file instead so that it's updated when there are changes to the subblock

drtrigon, 2019-11-25:
When placing (or may be when saving) the user should be able to choose between:
- place copy
- place ref
As booth behaviours have their advantages. I would not like to open a project after some months again in order to find it does not work anymore just because some other files have changed.

cavearr, 2020-01-26:
At present icestudio not support block reference but i'm labelling this issue as new feature becasuse it is a very interesting feature for next versions.

Thanks for your comments!