FPGAwars/icestudio #361: [BUG] View FPGAs resources

Obijuan, 2019-12-23:
In the current APIO/develop branch the arachne-pnr has been replace by **nextpnr**

The FPGAs resources were obtained by parsing the arachne-pnr output

As an example, this is an output for the ice40HX08 FPGA in the Alhambra-II board:

```
arachne-pnr -d 8k -P tq144:4k -p main.pcf -o hardware.asc hardware.blif
seed: 1
device: 8k
read_chipdb +/share/arachne-pnr/chipdb-8k.bin...
supported packages: bg121, bg121:4k, cb132, cb132:4k, cm121, cm121:4k, cm225, cm225:4k, cm81, cm81:4k, ct256, tq144:4k
read_blif hardware.blif...
prune...
read_pcf main.pcf...
instantiate_io...
pack...

After packing:
IOs          21 / 107
GBs          0 / 8
GB_IOs     0 / 8
LCs          229 / 7680
DFF        57
CARRY      16
CARRY, DFF 77
DFF PASS   26
CARRY PASS 12
BRAMs        0 / 32
WARMBOOTs    0 / 1
PLLs         0 / 2

place_constraints...
promote_globals...
promoted vclk$2, 134 / 134
promoted $abc$2777$n87, 19 / 20
promoted $abc$2777$n76, 19 / 20
promoted $abc$2777$n65, 19 / 20
promoted $abc$2777$n54, 19 / 20
promoted $abc$2777$n92, 15 / 15
promoted $abc$2777$n81, 15 / 15
promoted $abc$2777$n70, 15 / 15
promoted 8 nets
4 sr/we
3 cen/wclke
1 clk
8 globals
4 sr/we
3 cen/wclke
1 clk
realize_constants...
realized 1
place...
initial wire length = 4997
at iteration #50: temp = 15.9836, wire length = 3115
at iteration #100: temp = 9.56999, wire length = 2381
at iteration #150: temp = 4.43369, wire length = 1681
at iteration #200: temp = 1.58942, wire length = 1022
at iteration #250: temp = 0.0440565, wire length = 706
final wire length = 703

After placement:
PIOs       18 / 107
PLBs       79 / 960
BRAMs      0 / 32
```

Icestudio process that output and prints the following info in the bottom bar:

[image]

The new nextpnr produces the following output:

```
nextpnr-ice40 --hx8k --package tq144:4k --json hardware.json --asc hardware.asc --pcf main.pcf
Info: Importing module main
Info: Rule checker, verifying imported design
Info: Checksum: 0xa6568567

Info: constrained 'v901ecf[7]' to bel 'X4/Y0/io0'
Info: constrained 'v901ecf[6]' to bel 'X4/Y0/io1'
[...]

Info: Packing constants..
Info: Packing IOs..
Info: vcc0af0 feeds SB_IO vae165f.vcaef5f.v34955f.io_pin, removing $nextpnr_ibuf vcc0af0.
Info: v0a4578 feeds SB_IO vae165f.v8090fb.v34955f.io_pin, removing $nextpnr_ibuf v0a4578.
Info: v3c062f feeds SB_IO vae165f.va03463.v34955f.io_pin, removing $nextpnr_ibuf v3c062f.
Info: v087b32 feeds SB_IO vae165f.v92494d.v34955f.io_pin, removing $nextpnr_ibuf v087b32.
Info: Packing LUT-FFs..
Info:       65 LCs used as LUT4 only
Info:      108 LCs used as LUT4 and DFF
Info: Packing non-LUT FFs..
Info:       26 LCs used as DFF only
Info: Packing carries..
Info:        7 LCs used as CARRY only
Info: Packing RAMs..
Info: Placing PLLs..
Info: Packing special functions..
Info: Promoting globals..
Info: promoting vclk$SB_IO_IN (fanout 134)
[..]

Info: Device utilisation:
Info: 	         ICESTORM_LC:   216/ 7680     2%
Info: 	        ICESTORM_RAM:     0/   32     0%
Info: 	               SB_IO:    21/  256     8%
Info: 	               SB_GB:     8/    8   100%
Info: 	        ICESTORM_PLL:     0/    2     0%
Info: 	         SB_WARMBOOT:     0/    1     0%

Info: Placed 21 cells based on constraints.
Info: Creating initial analytic placement for 131 cells, random placement wirelen = 4365.
Info:     at initial placer iter 0, wirelen = 373
Info:     at initial placer iter 1, wirelen = 375
Info:     at initial placer iter 2, wirelen = 374
Info:     at initial placer iter 3, wirelen = 375
Info: Running main analytical placer.
Info:     at iteration #1, type ICESTORM_LC: wirelen solved = 378, spread = 682, legal = 787; time = 0.01s
Info:     at iteration #1, type SB_GB: wirelen solved = 776, spread = 954, legal = 1009; time = 0.01s
Info:     at iteration #1, type ALL: wirelen solved = 395, spread = 705, legal = 845; time = 0.01s
[...]
Info: Max frequency for clock 'vclk$SB_IO_IN_$glb_clk': 87.67 MHz (PASS at 12.00 MHz)
[...]
```

**My proposal** is to **only show** the information provided by the "Info: Device utilisation"

```
Info: Device utilisation:
Info: 	         ICESTORM_LC:   216/ 7680     2%
Info: 	        ICESTORM_RAM:     0/   32     0%
Info: 	               SB_IO:    21/  256     8%
Info: 	               SB_GB:     8/    8   100%
Info: 	        ICESTORM_PLL:     0/    2     0%
Info: 	         SB_WARMBOOT:     0/    1     0%
```

In the icestudio botom line the information could be displayed like this:

LC: 216/7680 (2%),  RAM: 0/32 (0%),  IO: 21/256 (8%),  GB: 21/256 (100%), PLL: 0/2 (0%), WB: 0/1 (0%)

If more information is needed, the user should consult the View command output

It would be **very nice** is the** Maximum frequency** is also shown. The line to parse is this one:

```
Info: Max frequency for clock 'vclk$SB_IO_IN_$glb_clk': 87.67 MHz (PASS at 12.00 MHz)
```
The output in the bottom line could be something like:

Max Freq: 87.67Mhz

For other FPGAs the nextpnr output will be different, with other resources

cavearr, 2019-12-23:
thanks @Obijuan! i'm working now on it!

cavearr, 2019-12-23:
update complete! now we have bottom bar with nextpnr information and Max Frequency.

I'm maintaining this issue open until we check all compatible FPGAs with this new output.

Obijuan, 2019-12-23:
Thanks!!!! :-)  
Tested with Alhambra-II: OK!

Obijuan, 2019-12-23:
Tested with:
* icezum Alhambra: OK!
* icestick: OK!
* Nandland go board: OK!

Obijuan, 2019-12-23:
Tested with: 
* Back ice: OK!
* TinyFPGA B2: OK!
* TinyFPGA BX: OK!

Obijuan, 2019-12-23:
Tested with:
* Blackice II: OK!
* BlackIce MX: OK!
* ice40-hx8k Breakout Board: OK!

Obijuan, 2019-12-23:
Tested with:

* Kéfir: OK!
* ICe40-UltraPlus Breakout Board: OK!
   This one have the up5K FPGA. It have more resources available as shown by the nextpnr output:
```
Info: Device utilisation:
Info: 	         ICESTORM_LC:   216/ 5280     4%
Info: 	        ICESTORM_RAM:     0/   30     0%
Info: 	               SB_IO:    13/   96    13%
Info: 	               SB_GB:     8/    8   100%
Info: 	        ICESTORM_PLL:     0/    1     0%
Info: 	         SB_WARMBOOT:     0/    1     0%
Info: 	        ICESTORM_DSP:     0/    8     0%
Info: 	      ICESTORM_HFOSC:     0/    1     0%
Info: 	      ICESTORM_LFOSC:     0/    1     0%
Info: 	              SB_I2C:     0/    2     0%
Info: 	              SB_SPI:     0/    2     0%
Info: 	              IO_I3C:     0/    2     0%
Info: 	         SB_LEDDA_IP:     0/    1     0%
Info: 	         SB_RGBA_DRV:     0/    1     0%
Info: 	      ICESTORM_SPRAM:     0/    4     0%
```

But the common resources are detected OK!  LC, RAM, IO, etc...

Obijuan, 2019-12-23:
Test with:

* Icebreaker: OK!
* Icesbreaker bitsy: OK!
* upduino v1.0: OK!
* upduino v2.0: OK!

Obijuan, 2019-12-23:
Tested with:

* FPGA 101 Workshop Badge Board: OK!

Obijuan, 2019-12-23:
ECP5 FPGAs:
* ULX3S-12F: The max frequency has been detected ok, but the output for the resources is different than for ice40 fpgas:

' ' ' 
nextpnr-ecp5 --25k --package CABGA381 --json hardware.json --textcfg hardware.config --lpf main.lpf --timing-allow-fail
Info: Importing module main
Info: Rule checker, verifying imported design
Info: Checksum: 0x9b21e64c

Info: Logic utilisation before packing:
Info:     Total LUT4s:       260/24288     1%
Info:         logic LUTs:    158/24288     0%
Info:         carry LUTs:    102/24288     0%
Info:           RAM LUTs:      0/12144     0%
Info:          RAMW LUTs:      0/ 6072     0%

Info:      Total DFFs:       268/24288     1%
Info: Packing IOs..
Info: pin 'vcc0af0$tr_io' constrained to Bel 'X4/Y50/PIOA'.
[...]
Info: pin 'v901ecf[7]$tr_io' constrained to Bel 'X0/Y20/PIOD'.
Info: Packing constants..
Info: Packing carries...
Info: Finding LUTFF pairs...
Info: Packing LUT5-7s...
Info: Finding LUT-LUT pairs...
Info: Packing paired LUTs into a SLICE...
Info: Packing unpaired LUTs into a SLICE...
Info: Packing unpaired FFs into a SLICE...
Info: Generating derived timing constraints...
Info: Promoting globals...
Info:     promoting clock net vclk$TRELLIS_IO_IN to global network
Info: Checksum: 0x44526653

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
Info: Checksum: 0x36487996

Info: Device utilisation:
Info: 	       TRELLIS_SLICE:   241/12144     1%
Info: 	          TRELLIS_IO:    13/  196     6%
Info: 	                DCCA:     1/   56     1%
Info: 	              DP16KD:     0/   56     0%
Info: 	          MULT18X18D:     0/   28     0%
Info: 	              ALU54B:     0/   14     0%
Info: 	             EHXPLLL:     0/    2     0%
Info: 	             EXTREFB:     0/    1     0%
Info: 	                DCUA:     0/    1     0%
Info: 	           PCSCLKDIV:     0/    2     0%
Info: 	             IOLOGIC:     0/  128     0%
Info: 	            SIOLOGIC:     0/   68     0%
Info: 	                 GSR:     0/    1     0%
Info: 	               JTAGG:     0/    1     0%
Info: 	                OSCG:     0/    1     0%
Info: 	               SEDGA:     0/    1     0%
Info: 	                 DTR:     0/    1     0%
Info: 	             USRMCLK:     0/    1     0%
Info: 	             CLKDIVF:     0/    4     0%
Info: 	           ECLKSYNCB:     0/   10     0%
Info: 	             DLLDELD:     0/    8     0%
Info: 	              DDRDLL:     0/    4     0%
Info: 	             DQSBUFM:     0/    8     0%
Info: 	     TRELLIS_ECLKBUF:     0/    8     0%
Info: 	        ECLKBRIDGECS:     0/    2     0%

Info: Placed 13 cells based on constraints.
[...]
' ' '

Obijuan, 2019-12-23:
* ULX3S-25F: Same result as ULX3S-12F
* ULS3S-45F: Same result as ULX3S-12F/25F
* ULS3S-85F: Same result as ULX3S-12F/25F/45F

The nextpnr output is similar to all these boards. Therefore, when it is working for one of them it will work for the rest

cavearr, 2020-01-26:
We change this issue to enhanced label and track de new improvements in this feature.

Obijuan, 2022-01-17:
I will close this issue, as in the latest icestudio (0.8.2w) the resources are alredy shown ok

closed 2022-01-17 by Obijuan