FPGAwars/icestudio #365: [BUG][Regression] The sintax error on the verilog blocks are not shown

Obijuan, 2019-12-24:
It is working ok in the 0.5.1 nightly, but it fails in the current develop branch

It can be reproduce just by adding a code block with syntax errors:

[image]

When the option Tools/Verify is click, the error is not reported

[image]

This is the information obtained in the Command ouptut windows:

```
export APIO_HOME_DIR=/home/obijuan/.icestudio/apio; "/home/obijuan/.icestudio/venv/bin/apio" verify --board alhambra-ii -p "/tmp/icestudio-30916V9E0v0Fo4i5x"

iverilog -B "/home/obijuan/.icestudio/apio/packages/toolchain-iverilog/lib/ivl" -o hardware.out -D VCD_OUTPUT= /home/obijuan/.icestudio/apio/packages/toolchain-yosys/share/yosys/ice40/cells_sim.v main.v
main.v:16: syntax error
I give up.
scons: *** [hardware.out] Error 2
========================= [ ERROR ] Took 0.28 seconds =========================
```

Obijuan, 2019-12-24:
This is the node version I am using for launching the icestudio
$ node --version
v12.14.0

cavearr, 2019-12-27:
Fixed! now when you create a new code block, icestudio needs almos one input or one output at code block creation.

closed 2019-12-27 by cavearr

Obijuan, 2019-12-28:
Great! It works now! :-)  Thanks!