FPGAwars/icestudio #367: [Help needed] Cannot build devel on macOS

nanoant, 2019-12-24:
I am having a trouble to build icestudio `devel` on macOS. First of all node-v12.14.0-darwin-x64 does not work at all. Some packages fail to build on `npm install`. When using node-v10.18.0-darwin-x64 it builds but when doing `npm run dist` I get `Apio settings are in debug mode` error. If I disable this check I get:
~~~
Fatal error: Command failed: cache/toolchain/venv/bin/pip download --dest cache/toolchain/default-apio apio[blackiceprog,tinyfpgab,tinyprog]">=0.5.0,<0.6.0"
~~~
I am definitely missing something. How do I pull latest `apio`?

cavearr, 2020-01-26:
Hi @nanoant ! Could you try with node 10.17.x version?

Obijuan, 2022-05-02:
Please, try it again with the latest icestudio version. I will close this issue. Please, feel free to reopen it if the problem remains

closed 2022-05-02 by Obijuan