FPGAwars/icestudio #368: Windows with Visual Studio Install Issues

gojimmypi, 2019-12-25:
I'm following up [my tweets](https://twitter.com/gojimmypi/status/1209494050367299586?s=20) with details of some of the problems I encountered during install of icestorm for the ULX3S on Windows 10.

First, it didn't work at all in WSL, I assume as Node version in WSL was 8 and version 12 was needed. `npm start` just sat waiting...

[image]

So then for a regular Windows install, I saw a ton of errors during `npm install`:

[image]

When I tried to ignore the errors and start icestudio anynow, all I saw was a blank Windows app:

[image]

I ended up rebooting... and unfortunately then lost the entire install history, so all that is left is that tweet pic for errors :/

After the reboot, a non-blank icestudio seems to have at least launched (I have no idea what underlying problems may exist due to errors during install)... however Python refused to be recognized, despite being able to go to a command prompt and type "python" and get to a python prompt... even after a new fresh install of a more recent Python.

As it turns out, there are a couple of script files such as [toolchain_builder.js](https://github.com/FPGAwars/icestudio/blob/dcbbbb7f9e28fd74b129a3d1ac67e85a3886d58a/tasks/toolchain_builder.js#L279) and [utils.js](https://github.com/FPGAwars/icestudio/blob/dcbbbb7f9e28fd74b129a3d1ac67e85a3886d58a/app/scripts/services/utils.js#L31) that have the `Python.exe` check hard coded like this:

```
    if (process.platform === 'win32') {
      possibleExecutables.push('python.exe');
      possibleExecutables.push('C:\\Python38\\python.exe');
      possibleExecutables.push('C:\\Python37\\python.exe');
      possibleExecutables.push('C:\\Python36\\python.exe');
      possibleExecutables.push('C:\\Python35\\python.exe');
    }
```
However, Python seems to recognize that I have Visual Studio installed, as my Python was in:

```
C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64
```
Even when re-installing, the updates did not go to any `C:\Python` directory.

I tried adding this line in hopes of solving this and submitting a PR:

```
      possibleExecutables.push('C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Python36_64\\python.exe');

```
but it was not successful. The icestudio app still did not find Python (which was essential to installing the ECP5 toolchain). I suspect there are more hard-coded patch checks in other places.

I don't have Python installed via Visual Studio 2019, 

[image]

but I *do* have it in the side-by-side installed of Visual Studio 2017:

[image]

(note that my Python is installed in the Visual Studio "share" directory.

I resorted to a brute-force solution of copying the contents of my

```
C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python36_64
```

to a new directory:

```
C:\Python36\
```

That seemed to resolve the Python issues for installing the toolchain! (yes, it takes some time to download)

As noted in the tweet thread, there were a few other non-intuitive problems, such as the sample code not building, but no error actually displayed in the little red/pink box:

[image]

@mmicko to the rescue here - I simply needed to change `gn27` to `led[0]` and it *WORKED!* :)

Having something like icestudio is really amazing, particularly for newbies like me just learning FPGAs. This is also really great for the [ULX3S](https://www.crowdsupply.com/radiona/ulx3s) that I hope to see more widely available soon.

Thank you all that contribute to these open source tools :)

Let me know how I can help. Cheers.

csylvain, 2020-01-15:
i installed the libusbK drivers with Zadig as instructed ("Interface 0" only) and the WTFpga board i have would not start up. i eventually ran Zadig again and replaced the windows driver on the second port ("Interface 1") and then the board would start up and both ports appear as "libusbK USB Devices" in devmgmt.msc

csylvain, 2020-01-15:
to install the toolchain, it was necessary for me to run "easy_install -U apio" manually from the directory where 'pip' is found (pip itself would throw an error on start up). having used 'easy_install' then ICEStudio could complete the toolchain installation

GnarlyDave, 2020-04-09:
I'm seeing some similar problems with a windows install and python detection.
I had multiple versions of python and VS installed but python didn't seem to be available. (I don't have %PYTHON% set)
Trying to install node.js and the development environment, suggests that node-gyp needs help to spot python:
CLI> npm config set python <path-to-python...>
find the path by using
 py --list-paths
which I was going to try adding to the nsis installer, and tweak it so it looks for >= python 3.7 versions instead of =3.7

cavearr, 2020-04-09:
Try to download the last nightly version and tell us!

GnarlyDave, 2020-04-09:
uninstall ice studio. uninstall python 2.7 (left 3.8.2 installed)
download nightly win 64
install says python 3.8.2 will be installed. despite already installed.

Ice studio then very slow to start, says toolchain broken click here....then complains that python 3.5 or better is needed.

cavearr, 2020-04-10:
We are working on this issue, the problem is that Icestudio don't recognize your python path. 

Don't worry we are improve this with a panel to manually configure your python path. 

Enviados: Jueves, 9 de Abril 2020 21:38:21

Obijuan, 2021-06-29:
It would be great if you could test the latest stable version: 0.6.0-rc2:

https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0-rc2

to check if the bug has been fixed or not. Any feedback is very welcome. Thanks! :-)

Obijuan, 2022-05-02:
As no feedback given since Jun29-2021, I assume that this bugs has been fixed in the latest icestudio versions.
Thanks for the feedback

closed 2022-05-02 by Obijuan