FPGAwars/icestudio #369: Build Error - Develop Branch

sclel016, 2019-12-25:
Hi,

I'm trying to build the project from source using the latest commit on develop branch with SHA-1 of 
`dcbbbb7f9e28fd74b129a3d1ac67e85a3886d58a`. I've included the error message I'm getting, is there any more information needed to help troubleshoot the issue?

```
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:219:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:274:12)
gyp ERR! System Linux 5.4.2-1-MANJARO
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/<user>/Downloads/icestudio/node_modules/iltorb
gyp ERR! node -v v13.3.0
gyp ERR! node-gyp -v v6.0.1
gyp ERR! not ok 
```

umarcor, 2019-12-25:
Icestudio seems to be tested (and probably built) with node v6 (see https://github.com/FPGAwars/icestudio/blob/develop/.travis.yml#L6). I have successfully built it with v11. However, I am not sure about v12 or v13. Moreover, I did need to use `yarn` instead of `npm`. `npm install` fails.

cavearr, 2020-01-02:
Hi! this is replied at #375
Thanks!

closed 2020-01-02 by cavearr