FPGAwars/icestudio #371: make designs board-agnostic

eine, 2019-12-25:
At now, when creating any design (even if it is blank) a target board needs to be selected. AFAIK, this is only used to fill two lists (names of top-level I/O 1 bit length prts); however, it forces designs/examples to be "*converted*" to target a different board.

IMHO, it would be preferred to make designs agnostic of the target board. Users would set unique output block names. Then, in a separate section/modal/menu, those output blocks can be assigned to one or multiple target boards. Ideally, it would be visualized as a table, so it'd be easy to spot differences/equivalencies between target boards.

Ref #247
Ref #276

/cc @esden

adumont, 2020-01-03:
Moreover, at the moment, many blocks in the collections have constants, for example in Timers, DIviders, ... usually they are used to divide the Board clock's frequency to adjust it in Hz or us/ms/sec. It would be great to have the board clock frequency defined in some board config panel, and not hardcode it in blocks. (That would suppose to modify all the blocks in collections, but that's a one time job)

cavearr, 2020-01-26:
Thanks for your ideas! It's very interesting for nexts versios of icestudio.

Thanks!