FPGAwars/icestudio #375: windows 10 installation errors

agnunez, 2019-12-29:
Trying to reinstall in windows 10 with node LTS 12.14 instaled by msi.
After cloning icestudio and develop branch, and issuing npm install it runs installing dependencies until throwing this error about a problem renaming file and missing dependency:

> icestudio@0.5.1 postinstall C:\Users\anunez\AppData\Roaming\npm\node_modules\icestudio
> npmpd && cd app && npm install && bower install && cd ../tasks && npm install && cd .. && grunt getcollection

No specific dependencies on this platform: win32
npm ERR! code ENOENT
npm ERR! syscall rename
npm ERR! path C:\Users\anunez\AppData\Roaming\npm\node_modules\.staging\icestudio-0072bb42\node_modules\adm-zip
npm ERR! dest C:\Users\anunez\AppData\Roaming\npm\node_modules\.staging\adm-zip-9f6e0076
npm ERR! errno -4058
npm ERR! enoent ENOENT: no such file or directory, rename 'C:\Users\anunez\AppData\Roaming\npm\node_modules\.staging\icestudio-0072bb42\node_modules\adm-zip' -> 'C:\Users\anunez\AppData\Roaming\npm\node_modules\.staging\adm-zip-9f6e0076'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\anunez\AppData\Roaming\npm-cache\_logs\2019-12-29T10_14_05_403Z-debug.log
npm WARN grunt-wget@0.1.3 requires a peer of grunt@~0.4.0 but none is installed. You must install peer dependencies yourself.

npm ERR! code ELIFECYCLE
npm ERR! errno 4294963238
npm ERR! icestudio@0.5.1 postinstall: `npmpd && cd app && npm install && bower install && cd ../tasks && npm install && cd .. && grunt getcollection`
npm ERR! Exit status 4294963238
npm ERR!
npm ERR! Failed at the icestudio@0.5.1 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\anunez\AppData\Roaming\npm-cache\_logs\2019-12-29T10_14_05_456Z-debug.log

cavearr, 2020-01-02:
Hi! this is an error in documentation, this week we are updating it.

To develop on windows you need to install nodejs version 10.17 , this is a problem between some node js components in windows with old libraries used in icestudio (we are planning a new icestudio version rewritting all of it with all learned until now).

If you install this nodejs version you can install without problems.

Thanks!

cavearr, 2020-01-02:
I'm closing this issue if you try and doesn't work reopen it please!

closed 2020-01-02 by cavearr