FPGAwars/icestudio #376: HiDPI support?

whitequark, 2020-01-05:
I have a Linux system and a high-density display:
```
screen #0:
  dimensions:    3200x1800 pixels (294x165 millimeters)
  resolution:    276x277 dots per inch
```
Everything in icestudio is *way* too small:
[image]

I can't seem to find any configuration for HiDPI nor does it pick up system settings. Is there just no HiDPI support?

cavearr, 2020-01-26:
i'm trying to support hidpi for next versions, but we doesn't have a displty to try it, we are fixing at soon as possible.

whitequark, 2020-01-27:
Got it. I'm happy to test it on my system if that helps.

lisovy, 2020-01-28:
I've just tested the latest version from git (commit 82697d29fcf8) and I can confirm it works on my 4k display without any issues -- I had the aforementioned issues with the stable 0.5.0.
I am using Ubuntu 19.10.

cavearr, 2025-01-24:
The last wip https://downloads.icestudio.io works very well in HiDPI displays,  Reopen if necesary!

closed 2025-01-24 by cavearr