FPGAwars/icestudio #377: FPGA I/O ports not defined

Luca8991, 2020-01-05:
I'm just trying to upload the _One LED_ Basic Example on my **TinyFPGA BX** board, but I have this error: `FPGA I/O ports not defined`

Here is the command output log:
```
arachne-pnr -d 8k -P cm81 -p main.pcf -o hardware.asc -q hardware.blif
main.pcf:4: fatal error: unknown pin `0' on package `cm81'
scons: *** [hardware.asc] Error 1
 [ ERROR ] Took 1.93 seconds
```

Obijuan, 2020-01-05:
The default examples are for the icezum Alhambra board. If you are using
another board you have to indicate in which pin the output should be
redirected

Just click on the output pull down menu and select the led you want to use
as output

Obijuan, 2020-01-05:
ok! I've reproduced the bug. The conversion between the pins does not work initially. You have to select another pin, and then the LED again, and it will work

It is shown in this gifs:

[image]

(I am using the build option for testing because I do not have the board here. You can use directly the upload option)

Obijuan, 2020-01-05:
@cavearr : I've also tested it in the development version and the bug is also there

Luca8991, 2020-01-05:
It works! Thank you very much @Obijuan!! 
Hope it'll be fixed soon

cavearr, 2020-01-26:
Hi @Luca8991 ! this is fixed now!  migration between boards is in very initial state, but we are improve¡ing with your tests! thanks for reporting!

FlyByPC, 2020-03-31:
@Obijuan Came here Googling this very bug. Switching to another pin and then back solved it. Thank you!

myzinsky, 2021-01-03:
Still a problem... just experienced it...

cavearr, 2025-01-19:
Hi! this bug is fixed for a long time ago, check the latest wips https://downloads.icestudio.io

closed 2025-01-19 by cavearr