FPGAwars/icestudio #379: Include kefír collection by default

Harvie, 2020-01-16:
I know it's mentioned in readme. But i think it's so interresting work it should come bundled to icestudio by default:
https://github.com/set-soft/collection-kefir_i

But english translation of the whole collection might be bit time consuming :-/

eine, 2020-01-16:
Alternatively, being able to define available collections as a list of git repo URLs would make it easier to have them updated/modified. Icestudio might reuse the syntax for npm/yarn packages; and it would allow to show when newer versions are available (a la [david-dm.org](https://david-dm.org/FPGAwars/icestudio?type=dev)).

juanmard, 2020-01-22:
Why is a comment deleted without reason?

cavearr, 2020-01-26:
We are working on block and collection manager, we put your collection in nexts versions of icestudio.

Thanks for your work!