FPGAwars/icestudio #382: Support VHDL with GHDL

eine, 2020-01-21:
A couple of days ago, @antonblanchard tweeted:

> https://twitter.com/antonblanchard/status/1218733857735725057
> If you've always wanted to blink an LED using VHDL and all Open Source tools, wait no more! Tristan has been ironing the last bugs out of GHDL synthesis, and it's rapidly improving. Thanks to the GHDL Docker project, you don't need to install anything.

Today:

> https://twitter.com/antonblanchard/status/1219448773333487616
> #Microwatt now works with ghdl/yosys/nextpnr! You can synthesize a 64 bit @OpenPOWERorg
 soft processor using all Open Source tools. Thanks to Tristan who did a bunch of work in ghdl. Details in the patch: https://github.com/antonblanchard/microwatt/commit/db937403ec9aff71ba8c17f68b62ea40a575f51c
> 
> Here's my OrangeCrab running ppc64le @micropython!
>
> [image]

Furthermore, the next release of GHDL is about to be published (*"before end of February"*), which will move synthesis features out of beta:

> ghdl/ghdl#1108
> For synthesis: this is the new major feature.

Now that using GHDL as a VHDL frontend for yosys is usable for not only toy examples, I believe it would be interesting to support VHDL in icestudio. Ideally, it would allow to have a "microwatt-FPGA" (a la [RISC-V-FPGA](https://github.com/Obijuan/RISC-V-FPGA), [Z80-FPGA](https://github.com/Obijuan/Z80-FPGA) or [SAP-1-FPGA](https://github.com/Obijuan/SAP-1-FPGA)) and/or to reuse many of the existing blocks.

There are currently a few projects of different complexity, which can be used for development/testing VHDL support in icestudio:

- https://github.com/tgingold/ghdlsynth-beta/tree/master/examples
- https://github.com/antonblanchard/ghdl-yosys-blink
- https://github.com/antonblanchard/microwatt

All of these are being tested in CI workflows with OCI images maintained in [ghdl/docker](https://github.com/ghdl/docker). Hence, this issue is partially related to #381. Nevertheless, it is also possible to add GHDL to apio (with some limitations on Windows).

/cc @adumont @andresdemski @juanmard @martinribelotta @RCoeurjoly @set-soft

Paebbels, 2020-01-21:
@Obijuan why was this issue closed without any reason?

Obijuan, 2020-01-21:
by mistake. Reopened

cavearr, 2020-01-26:
I'm labeling as enhancement for nexts icestudio versions.

mjcherbert, 2021-09-18:
is this feature approaching integration?  am keen for apio to know about vhdl with yosys+ghdl