FPGAwars/icestudio #383: JK with reset (verilog)

nanoolatz, 2020-01-29:
Hi all,
I am trying to add a Reset input to a JK flip-flop in verilog. The JK itself works properly, however when I've added the Reset input an error appears saying "errors detected in design"
I am new in verilog. Please find below the code that I am using
reg q = INI;
if (R == 1)
q <= 1'b0;
else
always @(posedge clk)
if (J == 0 && K == 1 )
q <= 1'b0;
else if (J == 1 && K == 1)
q <= ~q;
else if (J == 1 && K == 0)
q <= 1'b1;

Can you help me?

Thank you.

Olatz

cavearr, 2020-01-29:
Hi @nanoolatz ! sorry but this is not a proper place for verilog  questions.

If you want you can try  on our group that is a generic forum around open source fpgas:

https://groups.google.com/forum/#!forum/fpga-wars-explorando-el-lado-libre

thanks!

closed 2020-01-29 by cavearr

TimRudy, 2020-01-29:
However, two things:

If you get a generic message like "errors detected in design", you may be
able to learn more by looking at the actual output that came from the
toolchain, using View -> Command Output. (I would not hope too much for a
good explanation - Verilog simulation and synthesizing output is not very
helpful to the beginner!)

(Depending what version of Icestudio you have, there may be a bug where the
Command Output window does not receive the output as soon as you use Tools
-> Verify or Tools -> Build; and if that is the case, open a new Command
Output windows after you issue the command.)

Take a look at my implementation of JK flip-flop here:

https://github.com/TimRudy/ice-chips-verilog/blob/master/source-7400/7473.v

You should decide whether your Reset should be synchronous (rely on the
"posedge clk" in the always block), or asynchronous (add your "R" to the
sensitivity list, so that "R" itself can trigger it without clk changing).
In either case, your logic has to go inside the always block - never heard
of an "else" followed by an always block ;-) !

I would be careful if looking here:
https://github.com/TimRudy/ice-chips-verilog/blob/master/source-7400/74112.v

In the world of FPGAs, there should NOT be two asynchronous signals in the
sensitivity list (reset and set); therefore the 74112 code is due to be
changed soon to remove one of them (it's a long story).

Best of luck - Tim

TimRudy, 2020-01-29:
P.S. There is also a subtle issue with your code, when you are describing
hardware you must ask whether you're covering all the cases and whether you
have a final "else". In this code I do not see a case for "J == 0 && K ==
0". ...Right?! A missing logical case will generally lead to big problems.

if (J == 0 && K == 1 )
q <= 1'b0;
else if (J == 1 && K == 1)
q <= ~q;
else if (J == 1 && K == 0)
q <= 1'b1;

You can reach me offline, @ timrudy0@gmail.com. I didn't mention if you are
using my flip-flop code, you can remove the "BLOCKS" and the "generate" and
"[i]"... You don't need that distraction.

nanoolatz, 2020-01-31:
Thank you to all of you! Tim, I've managed to solve looking your example and introducing the Reset inside always. 

Thank you again!

Olatz

nanoolatz, 2020-01-31:
Solution 

reg q = INI;
always @(posedge clk or posedge Rst)
begin
    if (Rst==1)
        q <=1'b0;
    else
        if (J == 0 && K == 1 )
            q <= 1'b0;
        else if (J == 1 && K == 1)
            q <= ~q;
        else if (J == 1 && K == 0)
            q <= 1'b1;
end