FPGAwars/icestudio #384: Support "output reg" as output type for a module

adumont, 2020-01-29:
When importing Verilog code to a code block, sometimes a module can use "output reg" as output.

Importing such code can be quite painful when it could be almost straightforward if "output reg" was supported.

Example of such module : [ControlUnit.v](https://github.com/adumont/hrm-cpu/blob/39ace17c8582bbe958f326fe0e76cb07e74930b2/verilog/ControlUnit.v)

Idea of how to implement: 
- I guess adding a new text field asking "Enter the output reg ports", under "Enter the output ports" could work.
- Another way to be to let the user name the output port "reg portName", if portName is an output reg instead of an output wire.

Jesus89, 2020-01-30:
Hi @adumont, I think declaring the port `reg portName;` inside an Icestudio block should work.

However, it's interesting to support this explicitly in the UI. My suggestion is to add a boolean parameter in the output form like `[x] Register`, disabled by default to keep the same behavior.

Obijuan, 2020-01-30:
Yes, I use what @Jesus89 mention. Just adding reg portName; in the begining of the blocks  
The blocks I usually use are not so big, so I can refactor them very fast to fit in Icestudio blocks

If there is a very big component, you can always just instantiate the block directly from verilog (and do not have to change anything, just creating the icestudio block for the instantiation. This is a way of having both the icestudio blocks and the veriog files separated, but usable from both worlds

I do not use the latter approach because I want to give the user only one .ice with all the components embeded on it (I am focusing mainly on beginners and non-technical people) 

But if Icestudio implements the option Jesus mention it will be very welcome :-)

adumont, 2020-01-30:
Ah I didn't think of trying to add a " reg portname ; "line to the
beginning of the block. I'll try that. Thanks.

adumont, 2020-01-31:
If you happen to add this feature, you'll need to also consider "signed" reg/wire.

adumont, 2020-02-02:
For now, I've written a small python script to help me generate the list of parameters, input and output ports from a verilog module file. It also gives me all the necesary lines to add at the top (like reg signed [7:0] portname)... Useful if you have many ports...

I've explained it here: https://github.com/adumont/hrm-cpu/wiki/Instanciate-Verilog-submodule#help-with-instanciation-as-code-blocks-in-icestudio . 

I guess the funcionality could be added in IceSTudio. Especially I still miss the generation of each input/output ports "pins" if we are designing a sub-block... which is still a pain. I'll see if I can still do it similary, generating them in json and then I'll just have to copy/paste the json into the GUI... Next would be to connect the ports from the code block to each PIN block automatically ;).

TimRudy, 2020-10-15:
This is a good feature that could be integrated in Icestudio. Alex already did most of the leg work:
- Auto-populate the module ports dialog by parsing a Verilog code block that you want to import into Icestudio
- Or make a more direct UI "import" method:  
Use a file open dialog, or a "What do you want to import?" dialog that can accept Verilog pasted in. Then the module ports dialog, a "Register" boolean checkbox, and "signed", and doing outer input/output pin generation can become parts of that UI sequence