FPGAwars/icestudio #392: Add support for the Orange Crab

jmi2k, 2020-03-28:
I've backed this board, [the Orange Crab](https://github.com/gregdavill/OrangeCrab), based on the ECP5-25F, which would benefit greatly from Icestudio support. It features a Feather-compatible pinout, 1 Gbit of DRAM and a built-in ADC. It has even successfully booted Linux out of a Litex SoC RISC-V core.

Obijuan, 2020-04-09:
The first step is to support it in [Apio](https://github.com/FPGAwars/apio). Icestudio just call apio for performing the synthesis. I've opened [this issue](https://github.com/FPGAwars/apio/issues/198) in apio

Obijuan, 2020-04-24:
I've commited and initial support for the board. The .pcf file is missing.  @jmi2k do you know if this file is available or should we wait for it?

Obijuan, 2020-04-24:
ok, I've found a very simple .pcf file here: https://github.com/gregdavill/OrangeCrab-examples/tree/master/verilog/blink  It is not complete, just the clk and three leds but i can use it for the initial testings

Obijuan, 2020-04-24:
I've been able to synthesize this hello world example ok, but I cannot test on a real board (because I do not have an orangecrab board)

[image]

It would be great if you could test it (@jmi2k) . You have to use icestudio from the develop branch

jmi2k, 2020-04-24:
I don't have the board yet, I'm still awaiting for my group buy unit to come. Let's see if someone with a prototype can provide some help.

jmi2k, 2020-04-24:
I've updated the pinout.pcf following [this](https://gregdavill.github.io/OrangeCrab/r0.2/docs/pinout/) as reference, but I don't know how to generate the pinout.json (or even if it's required!). Can you help me?

Obijuan, 2020-04-25:
Yes! Great! could you emit a pull-req with the .pcf file? For generating the pinout.json there a script in the app/resources/boards/ called generator.py for converting from .pcf to .json... but I am not sure if it work with the new .pcf sintax (used by nextpnr). I have to try it and modify the script accordingly

daveshah1, 2020-04-25:
ECP5 does not use PCF files at all, it uses LPF files (whether with nextpnr or with Diamond).

jmi2k, 2020-04-25:
I've uploaded the PR, with the file extension changed to .lpf. I've followed the ULX3S pinout as reference, so the syntax should be right.

gregdavill, 2020-04-25:
Gave this a spin. It's not working exactly right. I can compile an example project, but loading it fails.

The main issue is that my PC's has an internal dfu device. So I get an error at the programmer saying that I need to specify a device.
```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device
```

This can be done in the following ways:
either add the device VID/PID pair to the dfu-util
```
dfu-util -d 1209:5af0 -D bitstream.bit
```
Or in the bitstream add a dfu-suffix
```
dfu-suffix -v 1209 -p 5af0 -a bitstream.bit
dfu-util -D bitstream.bit
```

The programmer is also currently using `sudo dfu-util`. It's probably better to add a udev rule to enable user access, instead of requiring sudo.

Obijuan, 2020-04-25:
> Gave this a spin. It's not working exactly right. I can compile an example project, but loading it fails

That should be solved in apio (which is the backend). Icestudio uses apio for building and uploading

I will re-open the issue in apio

Obijuan, 2020-04-25:
I've generted the pinout.json file. Now the pins are accessible from Icestudio. The bitstream generation is working fine

[image]