FPGAwars/icestudio #394: Nandland Go board doesn't build initially

uoo, 2020-04-03:
Running latest stable build (0.5.0) on MacOS.

If I select the Nandland Go board and make a minimal device (logic 1 going to LED1 as an output) and build, I get a "FPGA I/O ports not defined" error and in "view command output", I see:

main.pcf:3: fatal error: set_io: too few arguments
scons: *** [hardware.asc] Error 1

If I view the build directory, there are only 3 files:  hardware.blif, main.pcf, and main.v

However, if I select an Alhambra board, reset the output block to the Alhambra's LED1, and build, it builds fine.  At this point, the build directory contains new hardware.asc and hardware.bin files.

If I then select the Nandland Go board and reset the output block to the Go's LED1 and build, it works fine.

uoo, 2020-04-03:
Here's the view command output:

`[Fri Apr  3 08:24:40 2020] Processing go-board
--------------------------------------------------------------------------------
yosys -p "synth_ice40 -blif hardware.blif" -q main.v
arachne-pnr -d 1k -P vq100 -p main.pcf -o hardware.asc hardware.blif
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100
read_blif hardware.blif...
prune...
read_pcf main.pcf...
main.pcf:3: fatal error: set_io: too few arguments
scons: *** [hardware.asc] Error 1
========================= [ ERROR ] Took 0.87 seconds =========================
`

Obijuan, 2020-04-26:
I am working on it now. in the mean time  you can prevent this error by selecting a pin name when the output block is placed. For example chose LED2 (then you can change to LED1) and I will work

Obijuan, 2020-04-26:
Bug solved in the development version. Thanks for reporting! 🙂

[image]

closed 2020-04-26 by Obijuan