FPGAwars/icestudio #395: How to MacOSX

cavearr, 2020-04-03:
# This is not an issue, this is a short tutorial of how to install and run Icestudio on Mac OSX Mojave and Catalina.

Mac OSX Mojave and Catalina generate some errors in Icestudio.

In some cases, application crashes and others don't detect python correctly.

This is because it includes some security features that, for the moment, to run it, implies some tricky initial configuration.

First, you need to install python throught  [**brew**](https://brew.sh/index_es) to detect if correctly from Icestudio.

If you don't have brew installed, do:

1. Open terminal
2. Type:  **/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"**
3. Follow instructions

When finish type:

**brew install python**

The last sentence installs the last python3.X stable version at brew in your system (at April 4th of 2020, python 3.7) and this is compatible with others python installations that you have.

The final step to install and run Icestudio is disable security OSX Gatekeeper.

To do this, you need to type this into the terminal:

**sudo spctl --master-disable** 

Next **download the last NIGHTLY version** at:

[https://icestudio.io/](https://icestudio.io/)

Download the zip file, deflate and launch app (if you want you can drag into Applications folder or each of you want).

Once upon Icestudio is started, ask you for your board and ask you to install the toolchain.

When the toolchain process has been finished, Icestudio ask you for drivers install.

## Now you have your Icestudio ready to use!

Once you have toolchain and drivers installed, you could activate Gatekeeper again typing in the terminal:

**sudo spctl --master-enable**

If in the future you need to reinstall a new Icestudio fresh install or update the toolchain, you need to disable and reenable again.

It is possible that if you enable Gatekeeper, when you launch Icestudio first time after that, you need to go to **General Preferences -> Security** to enable it.

siliconwitch, 2020-09-12:
This doesn't seem to work anymore on Catalina 10.15.6. After disabling gatekeeper with `sudo spctl --master-disable` MacOS still complains that Python3 is damaged and won't run. There's no over-ride option shown either in `System Preferences -> Security & Privacy-> General`.

I tried this using the latest Python@3.8 from brew (specifically 3.8.5)

Also tried Python@3.7 but that didn't help. It's somehow created a new issue where once icestudio starts, the toolchain installer message "default toolchain not found ... requires internet connection. Do you want to continue" box now seems totally un-responsive. Pressing OK does nothing and the box just sits there. I can cancel and navigate around the rest of the app, but that ok button does nothing at all.

Going back to Python 3.8 had no effect, same issue. Re-installed Python from brew and removed the old one. Tried both pre-built nightly dmg, as well as running from source several times using `npm clean-install`.

The only thing I've noticed is when launching the prebuilt nwjs from terminal, I get the message `ERROR:web_contents_view_mac.mm(487)] Failed to look up NSViewBridgeFactoryHost!`. I'm not sure if this was there before or not, but it seems I've broken something.

siliconwitch, 2020-09-25:
The above didn't work for me on **Catalina 10.15.6** with **Python 3.8.5** installed from brew. I would still get the error that python is damaged and can't open, even after disabling macOS gatekeeper as shown above.

The solution was to fully turn off rootless aka System Integrity Protection (**SIP**).

1. Firstly, it's possible to get the application into a strange state if you've been messing around. Start by deleting the folders:
`~/.icestudio`
`~/Library/Application Support/icestudio`

2. Next **Restart** and hold **Cmd-R** until you reach the macOS recovery menu.

3. Along the top menu, select **Utilities** -> **Terminal** 

4. Disable SIP by typing `csrutil disable`

5. Reboot by typing `reboot`

6. Once restarted, you should be able to proceed with the install and no longer get the Python error.

7. Remember to go back into Recovery mode and enable SIP again once the toolchain is installed. Use the command `csrutil enable`

mSlider, 2022-01-08:
I'll put here some useful info, at least for macOS Sierra 10.12.6 

- icestudio .dmg installation with `csrutil disable` mentioned above.
- Once it opens install the latest toolchain if not installed before.
- Enable the drivers (this process in my case takes a lot: 8-10 minutes) but this is needed only the first time.
- Uploading a project would be pretty straight forward if it was not for the annoying `authentication panel` every time shows at the start and at the end of `uploading` process.

To solve this last problem, I've made a script to launch icestudio with admin privileges.
[https://github.com/mSlider/icestudio-macOS-launcher](url)

I hope someone find this useful.

Alex (mSlider)

jrwagz, 2022-11-09:
So, I've been trying to get this working on macOS Ventura 13.0 with a MacBook Air M2, 2022.

The main thing that doesn't seem to work properly (that I can find) is that the drivers don't install properly.

I'm using version `0.9.2w202211070711` and when installing the driver fails, I get this message: (yes it's repeated 7 times)

```
arch: Can't find brew in PATH arch: Can't find brew in PATH arch: Can't find brew in PATH arch: Can't find brew in PATH arch: Can't find brew in PATH arch: Can't find brew in PATH arch: Can't find brew in PATH
```

Since I know the app is running under rosetta, I tried to ensure that brew was also installed under rosetta, and it is, but for some reason it's not being found.

```
$ which brew
/opt/homebrew/bin/brew

$ arch -x86_64 /usr/local/bin/brew --version
Homebrew 3.6.9
Homebrew/homebrew-core (git revision 13c489653f4; last commit 2022-11-09)
$ arch -x86_64 /bin/bash -c "brew --version"
Homebrew 3.6.9
Homebrew/homebrew-core (git revision 7da925b66f7; last commit 2022-11-08)
Homebrew/homebrew-cask (git revision 8734006f21; last commit 2022-11-08)
```

So I have two versions of brew installed, `/opt/homebrew/bin/brew` is the native version, and `/usr/local/bin/brew` is the x86_64 version that runs under rosetta.  Perhaps I need to make sure that `/usr/local/bin` is in the PATH for icestudio?

cavearr, 2022-11-09:
Hi! try this solution and tell us if works for you:

https://stackoverflow.com/questions/64882584/how-to-run-the-homebrew-installer-under-rosetta-2-on-m1-macbook#:~:text=homebrew%2D3.0.0/)-,Share,-Improve%20this%20answer

jrwagz, 2022-11-09:
> Hi! try this solution and tell us if works for you:
> 
> 
> 
> https://stackoverflow.com/questions/64882584/how-to-run-the-homebrew-installer-under-rosetta-2-on-m1-macbook#:~:text=homebrew%2D3.0.0/)-,Share,-Improve%20this%20answer
> 
> 
> 
> 

Thanks for the pointer!  That's actually the exact same instructions I followed to get the x86_64 version of brew installed (as I showed in my last comment).  And with it installed I still get the same error that brew was not found in PATH.

jrwagz, 2022-11-09:
@cavearr , I found a solution to getting the drivers to install on apple silicon macs and submitted a PR here https://github.com/FPGAwars/icestudio/pull/660

cavearr, 2022-11-09:
Thanks a lot @jrwagz ! PR accepted and wips have been generated!

cavearr, 2024-02-05:
I'm closing this issue because i'm releasing a native osx arm 64 bundle.

This bundle not only fix the native support , the weid permissions problems are fixed, icestudio ask you for administrator password when it need it and apply the correct permissions.

Now the process has been simplified a lot.

If you need reopen the issue, do it!

closed 2024-02-05 by cavearr