FPGAwars/icestudio #397: Support for upduino 2.1?

wgaylord, 2020-04-09:
The current nightly build (Because I need the support for pre-installed python3.8) supports upduino 1 and 2. I have a upduino 2.1 I have tried to use the upduino 2 setting but icestudio can not detect it.

How would I go about adding support for this minor version change?

Obijuan, 2020-04-09:
As the upduino 2.1 is already supported by Apio, it should be straightforward in icestudio. I've added and experimental support, but as I do not have that board, I cannot test it.

Please, verify the information in the files in the folder: icestudio/app/resources/boards/upduino21. I've copied the files from the upduino 2.0, but I do not know if the .pcf is the same. It would be great if you could check it

For testing, you have use the development version (develop branch). Start icestudio and check the menu board: the upduino 2.1 should be there

wgaylord, 2020-04-10:
I am going to have to wait for the nightly build. Trying to install from github emits errors due to npm being unable to unlink files.

Obijuan, 2020-04-10:
It would be really nice if you can make it works. Waiting for the next build could take some time. So, let's first  fix this problem: making you able to execute icestudio development version

* Just for the record, Which is your operating system?

For executing icestudio from the repo you need to install node 10.17.0 (this is important). Then you have to execute the commands:

* npm install
* npm start

 In case of error, please, write here the output

Thanks a lot for your help

wgaylord, 2020-04-10:
Ah might be because I have the latest LTS (12.something) node installed. 

I am on windows btw.

Trying again with 10.17.0 installed.

wgaylord, 2020-04-10:
Okay while testing I can't flash to it. icestudio reports an error as "Unknown Board" Is there a debugging logger I can enable?

Obijuan, 2020-04-10:
Let's check your version. Open icestudio dev and go to Tools/Toolchains:

[image]

The latest version is Apio 0.5.3b0

If you are not using the latest version, click on Tools/Toolchain/update

Then select the upduino 2.1 board from Select/Board

[image]

You should see tha new board on the bottom right

[image]

Open a "Hello world" example and try to upload

Clicking on the View/Command Output you can see a log of what happended

wgaylord, 2020-04-10:
Hm Okay had to run the update.

Not sure what might be the problem. The PCF looks correct based off the information in the board's design repo.

> [Fri Apr 10 11:23:11 2020] Processing upduino21
> --------------------------------------------------------------------------------
> yosys -p "synth_ice40 -json hardware.json" -q main.v
> nextpnr-ice40 --up5k --package sg48 --json hardware.json --asc hardware.asc --pcf main.pcf -q
> ERROR: package does not have a pin named '0' (on line 3)
> ERROR: Loading PCF failed.
> 0 warnings, 2 errors
> scons: *** [hardware.asc] Error -1
> ========================= [ ERROR ] Took 1.82 seconds =========================
>

Obijuan, 2020-04-10:
I am trying to reproduce the error

You have updated the toolchain... can you tell me the version that appears  in your tool/toolchain option?  It should be  0.5.4 (the latest), but maybe there is another bug preventing it to update (i have to check that)

Obijuan, 2020-04-10:
OK. I think were is the problem. The support for the upduino 2.1 was added in Apio 0.5.4. But, due to a bug (i have to check), the Toolchain update is not working fine (At least in my computer. That is what I want you to tell me the version that has been installed when clicking on update)

wgaylord, 2020-04-10:
I have 0.5.4  now. It seems to be working now. (Was able to upload a constant lighting of an led) Thank you. I am going to try out some of the examples.

Obijuan, 2020-04-10:
Great!!! :-)

Did you install it manually? or by using the icestudio tool/update option?  I will leave this issue open until this bug was confirmed and fixed

wgaylord, 2020-04-10:
Just using the update option. I am having an issue trying to get stuff that requires a clock signal to work but that might just be that I don't know how to use icestudio yet.

Obijuan, 2020-04-10:
OK, I've fixed the problem with de update. Please, could you execute again Tool/Toolchain/UPdate?

Then, make sure the apio version is 0.5.4:

[image]

I want to make sure that the update was working ok, and that both icestudios, mine and yours, are equal. Then I will test some example for the upduino to reproduce your errors

Obijuan, 2020-04-10:
OK, The synthesis is working ok for me (i cannot upload because I do not have the board)

I am sending three examples. Use the option Tools/Builds to synthesize. send here the output of the View/Command Output option

[test-upduino.zip](https://github.com/FPGAwars/icestudio/files/4463310/test-upduino.zip)

wgaylord, 2020-04-11:
https://gist.github.com/wgaylord/a98abb120176bc3fe01a8c41bc01e881

All of them built and uploaded. They also all works. (First one had the LED  off because they use negative logic https://github.com/tinyvision-ai-inc/UPduino-v2.1/blob/master/RTL/src/upduino.pdc#L2)

Must just be me doing something wrong. Guess i have a lot to learn.

Obijuan, 2020-04-11:
Great! So i will close this issue as the upduino 2.1 is now supported in  icestudio-dev 

Thanks you very much for your help 🙂

closed 2020-04-11 by Obijuan

wgaylord, 2020-04-11:
Our welcome. Glad to help! Just need to figure out how to do simple stuff. (Like using T-Flip Flops and the such in this.) Is there any good tutorials that involve icestudio?

Obijuan, 2020-04-12:
I've written a [very basic tutorial](https://github.com/Obijuan/digital-electronics-with-open-FPGAs-tutorial/wiki), but it is in Spanish :-( I still did not have the time to translate into English:

You can also check the [Jedi Collection](https://github.com/FPGAwars/Collection-Jedi) for Icestudio, with lots of blocks. It is also in spanish mainly and the blocks has been tested in the Alhambra II board. But maybe it could help

ertgtct, 2020-05-13:
> https://gist.github.com/wgaylord/a98abb120176bc3fe01a8c41bc01e881
>
> All of them built and uploaded. They also all works. (First one had the LED off because they use negative logic [...]

You should not drive the RGB LED directly on the Board.
Here is a good tutorial [https://blog.idorobots.org/entries/upduino-fpga-tutorial.html](url) (it is for the UPduio v2.0, but should also work here).
You can implement the driver in an extra code Block, or you can use a premade Block with the driver implemented (attached to this post
[RGBdriver.zip](https://github.com/FPGAwars/icestudio/files/4622810/RGBdriver.zip)
).