FPGAwars/icestudio #398: icestudio (0.5.0, win64) hangs forever after launching

sebi5361, 2020-04-12:
When launching icestudio (0.5.0, win64), it hangs forever in that state:
[image]
What could it be? Thanks for your help.

cavearr, 2020-04-12:
Download the last nightly build, has better support for windows that 0.5.0 

You can download from https://icestudio.io 

Enviados: Domingo, 12 de Abril 2020 7:16:51 
Asunto: [FPGAwars/icestudio] icestudio (0.5.0, win64) hangs forever after launching (#398) 

[ https://user-images.githubusercontent.com/18078043/79061095-e1931580-7c8c-11ea-86fb-867f51caabd6.png ]

sebi5361, 2020-04-12:
I have just done it and I encounter the very same issue...
[image]

cavearr, 2020-04-12:
what is your windows version?

sebi5361, 2020-04-12:
Windows 10 Pro 64bit OS Build 19033.1 Does it help?
Is there a log file where I can track where it hangs?

sebi5361, 2020-04-12:
I made a minidump of icestudio while hanging, and analyzed it with WinDbg. Does it help?
```
EXCEPTION_RECORD:  (.exr -1)
ExceptionAddress: 0000000000000000
   ExceptionCode: 80000003 (Break instruction exception)
  ExceptionFlags: 00000000
NumberParameters: 0

FAULTING_THREAD:  000032d4

PROCESS_NAME:  icestudio.exe

ERROR_CODE: (NTSTATUS) 0x80000003 - {EXCEPTION}  Breakpoint  A breakpoint has been reached.

EXCEPTION_CODE_STR:  80000003

STACK_TEXT:  
00000000`005fedc8 00007ffc`56821eae : 00000000`005fee40 00007ffb`fcc1924e 00007ffc`01347ca0 00007ffb`fcbf22d8 : win32u!NtUserMsgWaitForMultipleObjectsEx+0x14
00000000`005fedd0 00007ffb`fcbd2837 : 402c0000`00000000 00000000`0278c3b8 0000f1ad`e53773ae 00000000`00000001 : user32!RealMsgWaitForMultipleObjectsEx+0x1e
00000000`005fee10 00007ffb`fcbd2608 : 00000000`00000001 00000000`00000000 00000000`0278c3b0 00007ffc`00237cb7 : nw!v8::internal::NativesCollection<2>::GetScriptsSource+0x4d82b7
00000000`005fef20 00007ffb`fcbd200e : 0000f1ad`e53770ce 00007ffc`012f56c0 00000000`005ff028 0000f1ad`e53770ee : nw!v8::internal::NativesCollection<2>::GetScriptsSource+0x4d8088
00000000`005fefd0 00007ffb`fcbf1681 : 00007ffb`fe9f5f88 00000000`027a9660 0000f1ad`e5376f4e 00000000`027ed5f8 : nw!v8::internal::NativesCollection<2>::GetScriptsSource+0x4d7a8e
00000000`005ff020 00007ffb`fddde1ee : 00000000`00000001 00007ffb`fcbdfe70 00000000`005ff158 00000000`00000000 : nw!v8::internal::NativesCollection<2>::GetScriptsSource+0x4f7101
00000000`005ff050 00007ffb`fb813274 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : nw!RelaunchChromeBrowserWithNewCommandLineIfNeeded+0x3475
00000000`005ff120 00007ffb`fb815cb1 : 00000000`027f8d00 00000000`005ff3a0 00000000`00000000 00000000`00e38100 : nw!v8::internal::compiler::MachineOperatorBuilder::Int32Mul+0x46994
00000000`005ff1e0 00007ffb`fb80fabe : 00000000`005ff2e8 00000000`005ff300 00000000`005ff7f8 00000000`00000000 : nw!v8::internal::compiler::MachineOperatorBuilder::Int32Mul+0x493d1
00000000`005ff210 00007ffb`fcb4f2c3 : 00000000`02795010 00007ffc`00237cb7 00000000`00000008 00000000`03938700 : nw!v8::internal::compiler::MachineOperatorBuilder::Int32Mul+0x431de
00000000`005ff2f0 00007ffb`fcb4fc92 : 00000000`005ff370 00007ffb`fada3348 00000000`00e30000 00000000`00e37910 : nw!v8::internal::NativesCollection<2>::GetScriptsSource+0x454d43
00000000`005ff360 00007ffb`fcb59c73 : 00000000`000000e4 00000000`02000002 00000000`00000003 00007ffb`fcc48d9b : nw!v8::internal::NativesCollection<2>::GetScriptsSource+0x455712
00000000`005ff430 00007ffb`fcb4f22a : 00000000`0069289a 00007ffb`fcbaaf15 00000000`0000002b 00000000`00e38060 : nw!v8::internal::NativesCollection<2>::GetScriptsSource+0x45f6f3
00000000`005ff7d0 00007ffb`fada11b0 : 00000000`005ff888 00000001`40000000 00000000`005ff9b0 00007ffc`56130885 : nw!v8::internal::NativesCollection<2>::GetScriptsSource+0x454caa
00000000`005ff860 00000001`400030dc : 00000000`00677df0 00007ffb`fada1098 00000000`005ff9d8 00000000`00677de0 : nw!ChromeMain+0x118
00000000`005ff950 00000001`400015ee : 00000000`00670150 00000000`00670150 00000000`0000000f 00000000`0000007f : icestudio+0x30dc
00000000`005ffa40 00000001`40120152 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : icestudio+0x15ee
00000000`005ffef0 00007ffc`57136fd4 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : icestudio+0x120152
00000000`005fff30 00007ffc`583bcf31 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : kernel32!BaseThreadInitThunk+0x14
00000000`005fff60 00000000`00000000 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x21

STACK_COMMAND:  ~0s; .ecxr ; kb

SYMBOL_NAME:  win32u!NtUserMsgWaitForMultipleObjectsEx+14

MODULE_NAME: win32u

IMAGE_NAME:  win32u.dll

FAILURE_BUCKET_ID:  BREAKPOINT_80000003_win32u.dll!NtUserMsgWaitForMultipleObjectsEx

OS_VERSION:  10.0.19033.1

BUILDLAB_STR:  vb_release

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {ab2d2e3e-edb3-a979-68f2-e3be023a5515}

Followup:     MachineOwner
---------

0:000> .exr -1
ExceptionAddress: 0000000000000000
   ExceptionCode: 80000003 (Break instruction exception)
  ExceptionFlags: 00000000
NumberParameters: 0

```

sebi5361, 2020-04-12:
I made some adjustments to my PATH, uninstalled and reinstalled icestudio and it works now.
I do not really understand why but it does...

closed 2020-04-12 by sebi5361

cavearr, 2020-04-13:
Hi sebi! thanks for your work! we are improving the  multiplatform installation to support cases like you with different standard paths or similar.

little by little the product will be better.

Thanks again !