FPGAwars/icestudio #400: warning to console

GnarlyDave, 2020-04-17:
if $HOME/.icestudio exists, but 
$HOME/.icestudio/collections doesn't exist

utils.js:439 Error: ENOENT: no such file or directory, scandir 'C:\Users\Dave\.icestudio\collections'
    at Object.readdirSync (fs.js:808)
    at Object.findCollections (utils.js:432)
    at Object.loadInternalCollections (collections.js:28)
    at Object.loadAllCollections (collections.js:15)
    at app.js:46
    at utils.js:1247
    at profile.js:45

I'd fix it myself, but it's 20 years since I did server side javascript and I'm still learning node.js :)

cavearr, 2020-04-18:
Thanks! Today i’m fixing it!

cavearr, 2020-11-10:
Now it's fixed on development branch, sorry for the delay!

closed 2020-11-10 by cavearr