FPGAwars/icestudio #402: Build fails if design contains default output

jdeitmerg, 2020-04-24:
Hi,

when adding a Basic->output FPGA pin to the design and **not** changing the pin in the dropdown, the build will fail. It will work once after temporarily selecting  a different pin in the dropdown.

This will be especially frustrating for beginners with a TinyFPGA BX, as the default output is the LED.
Also, the error popup is empty (this was different some versions ago):

[image]

apio command output:
```
set APIO_HOME_DIR=C:\Users\jonas\.icestudio\apio& "C:\Users\jonas\.icestudio\venv\Scripts\apio.exe" build --board TinyFPGA-BX -p "C:\Users\jonas\AppData\Local\Temp\icestudio-4996RrMLRDw8Owu1"

[Fri Apr 24 10:30:53 2020] Processing TinyFPGA-BX
--------------------------------------------------------------------------------
yosys -p "synth_ice40 -json hardware.json" -q main.v
nextpnr-ice40 --lp8k --package cm81 --json hardware.json --asc hardware.asc --pcf main.pcf -q
ERROR: expected PCF syntax 'set_io cell pin' (on line 3)
ERROR: Loading PCF failed.
0 warnings, 2 errors
scons: *** [hardware.asc] Error -1
========================= [ ERROR ] Took 4.80 seconds =========================
```

I'm on Version 0.5.1n200403, Apio 0.5.4

Obijuan, 2020-04-25:
Thanks for the report!  You are right. This is very bad for beginners (which are the main target for icestudio)

As a quick and dirty fix I've changed the initial pin name to NULL, so that you are forced to change it to the pin you want

The bug on error reporting is still there. I am working on that

[image]

Obijuan, 2020-04-26:
Bug fixed! (in the development version)

[image]

closed 2020-04-26 by Obijuan

jdeitmerg, 2020-04-27:
Thanks for the fix. Awesome response time!