FPGAwars/icestudio #406: ftdi_usb_get_strings failed en Linux Mint

migrassi, 2020-05-08:
Hola. Gran trabajo el que hacen con el Icestudio y Love Controls. Quiero trasladarlo a mis alumnos de Robótica en la Maestría en Artes Electrónicas de la Universidad Nacional de Tres de Febrero, en Buenos Aires, Argentina. Estoy teniendo un pequeño problema a la hora de conectar la placa (iCE40-HX8K Breadbord) con el icestudio para hacer las primeras pruebas. Obtengo el mensaje  ftdi_usb_get_strings failed: -4 (libusb_open() failed).

Todo lo que encuentro en el sitio relacionado con este problema se refiere a Windows y no a Linux, como es mi caso (Linux Mint 18.3 [sylvia]). Mucho agradeceré si me pueden comentar donde puedo encontrar información sobre este problema.

Muchas gracias y saludos desde Buenos Aires,

Miguel

Obijuan, 2020-05-09:
Hola!  Este tipo de cuestiones es mejor resolverlas en el grupo de FPGAwars. Ahí hay más gente que puede ayudar

https://groups.google.com/forum/#!forum/fpga-wars-explorando-el-lado-libre

En cualquier caso, te doy por aquí algunas pautas para saber qué está pasando. Tiene toda la pinta de ser un problema de permisos, pero hay que recolectar información primero de tu sistema

Vamos a comprobar si se te han instalado los ficheros correctamente. Abre un terminal y prueba ejecuta este comando::

```
$ ls /etc/udev/rules.d/
40-dfuse.rules      80-fpga-ftdi.rules    90-extraacl.rules  99-arduino-101.rules
70-snap.core.rules  80-fpga-serial.rules  98-openocd.rules   avrisp.rules
```
Comprueba que los ficheros 80-fpga-* aparecen. Estos son donde se configura a qué elementos del USB tienes acceso sin ser root. Se te deberían haber instalado al activar los drivers desde Icestudio

Comprueba también que pertecenes al grupo plugdev. Ejecuta el comando groups y comprueba que plugdev aparece:

```
$ groups 
obijuan adm tty uucp dialout cdrom sudo dip plugdev input lpadmin sambashare
```
Según los resultados de estas pruebas vemos qué otras cosas probar

migrassi, 2020-05-09:
Muchas gracias Obijuan !! Efectivamente, agregar mi usuario al grupo plugdev solucionó el problema. Si alguien tiene el mismo problema, cabe mencionar que se hace desde la consola con `sudo usermod -a -G plugdev usuario` y es necesario reiniciar (o actualizar de alguna otra forma que no conozco) para que IceStudio detecte el cambio (tipeando `group` en la consola tampoco aparece el grupo plugdev hasta que no se reinicia). Por cierto, cabe mencionar que en Windows 7 la instalación también fue muy simple y arrancó a la primera. Muchas gracias y que las FPGA libres sean contigo!!

Obijuan, 2020-05-12:
¡Muchas gracias! :smile:  
He añadido [esta nota](https://github.com/Obijuan/digital-electronics-with-open-FPGAs-tutorial/wiki/V%C3%ADdeo-2:-%C2%A1Poniendo-en-marcha-Icestudio!#nota-sobre-linux-mint) en el tutorial de instalación de icestudio
Cierro esta issue

closed 2020-05-12 by Obijuan