FPGAwars/icestudio #408: Support for internal Oscillator as Clock (UPduino v1 & v2)

ertgtct, 2020-05-13:
It would be great to be able to define one of the internal oscillators (the HF one in the best case) as input for the clock network. Then you dont have to route the clock discretely.
An example can be found here: [https://github.com/Idorobots/upduino-blinky/blob/master/blinky-extended/blinky.v](url) (Credits go to Idorobots for the great example).

dabrams, 2020-06-13:
I agree.  The verilog code provided by lattice is

	wire clk_48;
	SB_HFOSC u_hfosc (
		.CLKHFPU(1'b1),
		.CLKHFEN(1'b1),
		.CLKHF(clk_48)
	);

See https://github.com/FPGAwars/apio/issues/175

Laserlicht, 2020-06-27:
I agree too.

A solution could be an "output-clk". Then you only have to use the code-block above once. It would be possible the prescale the 48Mhz to the usual 12Mhz for compatibility reasons.

jrwagz, 2022-11-12:
I'll add my $0.02 here in saying this would be nice to have.

Idea: Is it possible to make your own custom block that you define the RTL for?  In that case, it is just as simple as putting this RTL in that block?

```
   wire clk;
   SB_HFOSC inthosc(.CLKHFPU(1'b1), .CLKHFEN(1'b1), .CLKHF(clk));
```

Has anyone tried this already?

jrwagz, 2022-11-12:
I just tested the above method, and it works.

[image]

Here is a link to the .ice file for this to use as a starter point:

https://gist.github.com/jrwagz/a2722055eea4211acd902e7548ee3905

cavearr, 2025-01-09:
Hi, i'm closing , if is needed , reopen it.

closed 2025-01-09 by cavearr