FPGAwars/icestudio #416: Loading indefinitely.

Paukler, 2020-05-31:
[image]
Newest verion of nightly build icestudio (0.5.1-n200523).
Newest version of python3 installed (3.8.3).
Newest version of nodejs(npm) installed (6.14.5).
Windows 10 Pro (Verion 1903, OS Build 18362.836).
Same directory works on different PC.
Probably same problem as #398 .

I tried to debug it myself but there are no Logs or anything.
```
PS F:\Electronics\FPGA\icestudio> npm start

> icestudio@0.5.1-n200523 start F:\Electronics\FPGA\icestudio
> grunt serve

Running "nggettext_compile:all" (nggettext_compile) task

Running "watch:scripts" (watch) task
Waiting...
Running "wiredep:task" (wiredep) task

Running "exec:stopNW" (exec) task
```

Paukler, 2020-05-31:
I just noticed both #398 and me have Windows Pro just wanted to point that out maybe there is a problem there?

cavearr, 2020-05-31:
I'm trying to send you an script to display more debug info, take me some time. 

Enviados: Domingo, 31 de Mayo 2020 20:20:52

Paukler, 2020-06-07:
I tried the build by umarcor just resulted in an additional window.
[image]
Nothing else changed.

cavearr, 2020-06-08:
Hi! i think is because you use nw.js for other project with newer version of it. The version used in icestudio is something old, ando probably the profile and configuration options are shared between projects and creating a conflit. 

Enviados: Domingo, 7 de Junio 2020 18:48:03 

[ https://user-images.githubusercontent.com/25936043/83974665-57afa380-a8ef-11ea-8f1b-91623dfae240.PNG ]

juanmard, 2020-06-08:
Hi @Paukler, I have the same message.
But now it works.
Just close the message, kill the "nwjs" processes and run it again.
I did not have to restart and it worked for me.

Paukler, 2020-06-08:
@cavearr I just installed the newest version of nw.js and don't use it for other projects. Wich version should i install?
@juanmard What exactly did you do and on what version since it still doesn't work for me.

juanmard, 2020-06-08:
Hi @Paukler ,

My downloaded version is: _**icestudio-0.5.1dev-win64**_ in a zip file. 

The message, as you can read, is a conflict of versions of _"NW.js"_, between the one you have in your user profile and the one used by _icestudio_.

To remove any possible previous version, the fastest is to delete the _"icestudio"_ folder in **%LocalAppData%**, then run _"icestudio.exe"_ again from your decompression folder.
After a few seconds, a new profile will be generated with the appropriate versions.

Regards

Paukler, 2020-06-08:
Hi @juanmard ,

Thank you the message is gone now but it still loads endlessly,
so we are back to 1² :/

juanmard, 2020-06-09:
A pity... :(((
As a reference, my SO: **Windows 10 Pro N - 1909** and works ok.

cavearr, 2020-06-09:
Hi Paukler, when start icestudio and loader freeze, click over loader (to obtain window focus) and press F12.

A new window appear, click con "Console option" and send us an screenshot.

Paukler, 2020-06-11:
[image]

cavearr, 2020-06-12:
Try the next steps: 

1) Uninstall Icestudio 
2) Uninstall nwjs , if you installed it and not use for nothing. 
3) Download and install the last nightly: [ https://github.com/cavearr/icestudio-nightly/raw/master/0.5.1n200523/icestudio-0.5.1n200523-win64.exe | https://github.com/cavearr/icestudio-nightly/raw/master/0.5.1n200523/icestudio-0.5.1n200523-win64.exe ] 

I think the problem is that you have installed nwjs in the last version that it is incompatible with icestudio (the path error is a false positive bug, a collateralbug that is not the real problem), 

Icestudio takes your nwjs instead the embeded and crash. 

You dont need to install nwjs to run icestudio, uninstall it run icestudio, should be work. 

Enviados: Jueves, 11 de Junio 2020 19:50:39 

[ https://user-images.githubusercontent.com/25936043/84422060-c0af5800-ac1c-11ea-8fce-17cd31212f3d.png ]

TurBoss, 2020-08-07:
Hello, 
I'm having this issue also but on linux
one of my project file is unable to load it takes forever
while the examples load fine

I attach the project file
[595.zip](https://github.com/FPGAwars/icestudio/files/5040833/595.zip)

Thanks

TurBoss, 2020-08-07:
oops sorry edit doesn't let me add new attachments
i uploaded a wrong file

here is the full file:
https://dpaste.com/9S6C3N9F2

TimRudy, 2020-08-08:
TurBoss thanks for example.

At joint.shapes.js:1451: `var radix = data.format;`
The radix is null and this causes crash at joint.shapes.js:1457: `var text = row.toString(radix).toUpperCase();`
"`RangeError: toString() radix argument must be between 2 and 36`"

The "data" was from a ROM block:
```
format: null
local: false
name: "ROM"
```
  with 16 bytes of data in its "list".

@cavearr do you think it should be fixed by validation before it gets here? Is ROM a special case that needs a fix?
Or should it be fixed at code level such as `var radix = data.format || 16;` or `var radix = data.format || 10;`?

cavearr, 2020-08-08:
Hi! On monday i will merge the solution. I am working on it!

----- Mensaje original -----

https://github.com/FPGAwars/icestudio/issues/416#issuecomment-670963755

cavearr, 2020-08-10:
Hi Tim! if you want , fix it with your solution, and do a pull request. I'm investigating this error but probably is a special case. For the moment with your validation code is enough.

TimRudy, 2020-08-11:
So the end-user problem can be solved by searching your .ice file for phrase `"format": null` as seen here right in the example file:
	"type": "basic.memory",
	"data": {
		"name": "ROM",
		"list": "FC\n60\nDA\nF2\n66\nB6\nBE\nE0\nFE\nE6\nFC\n60\nDA\nF2\n66\nB6",
		"local": false,
		"format": null
	}
That can't be a null. Change it to a 16 (no quotes, just a number) for hex, or 10 for decimal or 2 for binary, and what it represents is that the dropdown "Address format" in ROM block was unselected. Voila, now you'll have a dropdown selection.

TurBoss, 2020-08-11:
> So the end-user problem can be solved by searching your .ice file for phrase `"format": null` as seen here right in the example file:
> "type": "basic.memory",
> "data": {
> "name": "ROM",
> "list": [...]

just did that and project loaded fine

Thank you so much!

Paukler, 2020-09-08:
Sorry that i didn't reply for so long but life just hit and I kind of forgot about it.
Well whatever.

The problem still remains i am on the newest nightly and have no nwjs on my system anywhere from what i know.
And it still loads infinitely.

cavearr, 2021-01-07:
try with the new nightly 0.5.1n210107 https://downloads.icestudio.io

qpeters, 2021-07-13:
@Paukler is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

Paukler, 2021-07-15:
It seems to work with v0.6.0 now so I am gonna close this issue