FPGAwars/icestudio #417: Cannot find iceFUN in board select list but it is supported by apio

dabrams, 2020-06-04:
The iceFUN FPGA board is shown as a supported HX8K board in the Supported Boards list but it does not appear in the Select/Board list (Ver 0.5.1n200403).

I have the latest apio (Ver 0.4.1)  installed and it also shows the iceFUN supported.

Obijuan, 2020-06-04:
Please, update to the latest icestudio (icestudio-nightly 0.5.1-n200403). You can download it from here:

https://icestudio.io/#lk-download

This is an example of synthesizing a hello world circuit (just turning on a LED) in the IceFun Board (The board is available from the Selec/board menu)

[image]

Also make you have update the toolchain to the latest version (Apio 0.5.4):

[image]

I will close this issue

dabrams, 2020-06-04:
As I said in my post I **am** running the nightly build and I **have** updated to apio 0.5.4 and iceFUN is **NOT** appearing so I do not think the issue is closed yet.

[image]

[image]

[image]

Obijuan, 2020-06-04:
I am sorry @dabrams I did not see the icestudio version I just read the apio 0.4.1 and answered you. Thanks for the additional information

The board is implemented in the current icestudio develop branch (That's the one I am using). I did not realized that the board was not implemented in the previous version 0.5.1-n200403 (I got confused with other thread. I apollogize)

Either try it from the develop version or wait a week or so to the next nightly-release). I will leave the issue open until you confirm it is working

dabrams, 2020-06-04:
Thanks.  I had considered the TinyFPGA BX but prefer the iceFUN because of all the LEDs and pushbuttons. (I want to use it to teach a class and not having to hook up extra I/O makes it much easier.)

dabrams, 2020-06-09:
Thanks.  I am getting the following error.  I tired moving icefunprog.exe into the same directory as icestudio.exe but that did not work.  I also tried to add it to the path but that also did not work.  Where should the upload utility be located?  

Also, its there a file I can manually upload to the board?  I could not find the working directory for my project.

[image]

Jesus89, 2020-06-09:
You can export the Binary file and then uploaded it manually.

It seems that "icefunprog" was included as a [Python programmer in Apio](https://github.com/FPGAwars/apio/blob/dd3264a39e1cecf01aec437f97c8a412bb04a153/apio/resources/boards.json#L159), but it's not the same as, for example, tinyfpgab which is a real Python package installable via PyPI.

The proper way to distribute [iceFUNprog](https://github.com/devantech/iceFUNprog) is to prebuild packages, as we do for icestorm, iverilog, etc... Take this as a reference: https://github.com/FPGAwars/toolchain-icestorm.

The path is to cross-compile and distribute the programmer statically for different architectures in tar.gz files, [an example](https://github.com/FPGAwars/toolchain-icestorm/releases/tag/v1.11.1). Then, configure apio to be able to install this package: https://github.com/FPGAwars/apio/blob/master/apio/resources/distribution.json. And finally, it will be installed and included automatically in every Icestudio build to be used.

qpeters, 2021-07-13:
@dabrams is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

dabrams, 2021-07-18:
It works great.  I did a clean install on a spare machine and was able to create and download with no problems.

 

David

closed 2021-07-18 by Obijuan