FPGAwars/icestudio #420: TinyFPGA fails to build in IceStudio

unprovable, 2020-06-09:
Hello,

So, the main release, as well as the nightly, both fail to build for the TinyFPGA BX that I have. This works fine for 0.4.0, but not for any 0.5.0 build I have tried. Here's the error on the latest nightly - 
```
[Tue Jun  9 11:17:02 2020] Processing TinyFPGA-BX
--------------------------------------------------------------------------------
yosys -p "synth_ice40 -json hardware.json" -q main.v
main.v:74: Warning: Range [-1:-1] select out of bounds on signal `\divcounter': Setting 1 LSB bits to undef.
nextpnr-ice40 --lp8k --package cm81 --json hardware.json --asc hardware.asc --pcf main.pcf
Info: Importing module main
Info: Rule checker, verifying imported design
Info: Checksum: 0x88fa7f34

ERROR: expected PCF syntax 'set_io cell pin' (on line 3)
ERROR: Loading PCF failed.
0 warnings, 2 errors
scons: *** [hardware.asc] Error 255
========================= [ ERROR ] Took 3.21 seconds =========================
```

Also, worth noting that in any version of IceStorm, when you convert an example to TinyFPGA, it fails to bring with it the JSON pinout file from the PCF (or something like this), but this is remedied by building from scratch on 0.4.0 etc.

Cheers, M.

qpeters, 2021-07-13:
@unprovable is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

Obijuan, 2022-05-02:
Could you please try in recent version of icestudio? You can download the latest wip from here: https://downloads.icestudio.io/
I will close this issue. Feel free to reopen it if the same error remains (or open a new issue in case of new errors)
Thanks for the feedback

closed 2022-05-02 by Obijuan