FPGAwars/icestudio #422: trouble including verilog files

ke10g, 2020-06-29:
Hi. I'm trying to include some Verilog files (a library) in my design. 

I've made a header file with .vh suffix.
In this header file is a call to include the other files I want to include. 

but I get: ./myHeader.vh:6: Include file myLib.v not found. 

The files are all in the same directory as the icestudio file.

Obijuan, 2020-06-30:
Inside the code blocks, you have to indicate ALL the files that are part of your project (not only the top file, but all the files that belongs to your project)

So, you should also add this line:

//  @include myLib.v

This is not for including myLib.v into the verilog code (that is already done in myHeader.vh), but to indicate icestudio that myLib.v belongs to the same project

In this image you can see what should be put in the beginning of a code block that is created from 3 files: a verilog file, a verilog header and a memory file. From that point icestudio knows that the project consist of these three external files (and no more). The files not mentioned like this are ignored

[image]

ke10g, 2020-06-30:
Hi. Thanks for the quick response. I've done what you suggest and no longer get the error. 

But I'm confused as to how to call the functionality of the modules in myLib.v within one of my code blocks in my Icestudio module. 

In myLib.v I have declared a module with some inputs and outputs and internal functionality. How do I pass my signals to and from this module? Can you give an example of what I need to do within Icestudio to call this module?

Obijuan, 2020-06-30:
Working Inside the code blocks is the same than working in a verilog module: you have to instantiate the module you want to use. 

This is a hello world example composed of two files: the icestudio project and inv.v which contains a verilog module that is a not gate. I've written the content of the inv.v file in a comment block

[image]

The icestudio code block instantiate the inv module and connect it to its input (i) and to its output (o)

This is the whole project:

[verilog-test.zip](https://github.com/FPGAwars/icestudio/files/4851535/verilog-test.zip)

ke10g, 2020-06-30:
Thank you Juan. All working now.

Obijuan, 2020-06-30:
Great! I will close the issue

closed 2020-06-30 by Obijuan