FPGAwars/icestudio #423: Help debugging: Problem building two modules when connected.

ke10g, 2020-07-02:
I have two modules in a project that interact with each other with various wires and busses. These synthesize just fine (using 70 percent of IceStorm LC resources). But then I make a connection in the first module to the signal coming from the second module ( when I read the value from the bus), build fails yosys Device Utilization telling me that 533% of available IceStorm LCs are needed. 

Conceptually, I'm having trouble grasping what could be going wrong. Why would it make such a difference? The two modules work fine on their own, and generate the variables they are supposed to, but when connected at the end of the chain, they all of a sudden use up 6 times the resources. 

I wonder if any of you have ideas what I might look out for? They are running on the same clock, but maybe there is something to look out for timing wise? What else could be the issue?

Obijuan, 2020-07-05:
Please, include a .zip file so that we can have a look. It does not seem a problem with icestudio (icestudio is the front end. It generates the verilog code and pases to yosys). Very likely you are not implementing correctly the RAMs, but as we do not have your project we do not have enough information for helping you
When you modules are not connected, there are some unused parts inside that are removed by yosys (it is an optimization). When the modules are connected, these parts are not removed and you can see a big increase in the resources. But I am just guessing

ke10g, 2020-07-05:
Hi Juan, yes you are right. I figured it out after all. Things broke down when I connected the two modules because I was then effectuating read and write to the array in an uncoordinated manner and crossing clock domains. It started working when I instantiated a dual port ram module with appropriate clocking for reading and writing. Thanks!

Obijuan, 2020-07-06:
ok, I will close this issue then

closed 2020-07-06 by Obijuan