FPGAwars/icestudio #424: yosys not inferring array as block ram

ke10g, 2020-07-03:
Continued from : [](https://github.com/FPGAwars/icestudio/issues/423)

I now understand that the issue arises because when my modules are connected, specifically, when I have a wire sending values sequentially from an array to another module, this array gets inferred as logic. If I disconnect the modules they build just fine. 

I've tried dumping my array values to registers on all clock counts, I've tried separating always blocks between reading and writing blocks, I've tried writing on negedge and reading on posedge, etc... What else can I try? 

I am asking here because I am not getting help on other forums... so maybe it is IceStudio that is doing something strange. 

Please any help or hits appreciated.

Obijuan, 2020-07-05:
Please, include the exact example you are testing. It is not a problem with icestudio, but rather on how you are describing the arrays (memories). If they are not described correctly, yosys will not infer memory from them. It is also important that you provide information of the FPGA and board you are using

Obijuan, 2020-07-06:
solved (see [#423](https://github.com/FPGAwars/icestudio/issues/423))

closed 2020-07-06 by Obijuan