FPGAwars/icestudio #425: How to modify the thumbnail of a block?

AbsorbedInThought, 2020-07-17:
I have been trying to modify the thumbnail of a simple block such as in [AND.ice](https://github.com/FPGAwars/collection-default/blob/master/blocks/Logic/Gate/AND.ice). But the path for the image looks something like this:

```
"image": "%3Csvg%20xmlns=%22http://www.w3.org/2000/svg%22%20viewBox=%22-252%20400.9%2090%2040%22%3E%3Cpath%20d=%22M-252%20409.9h26v2h-26zM-252%20429.9h27v2h-27z%22/%3E%3Cpath%20d=%22M-227%20400.9v39.9h20.4c11.3%200%2020-9%2020-20s-8.7-20-20-20H-227zm2.9%202.8h17.6c9.8%200%2016.7%207.6%2016.7%2017.1%200%209.5-7.4%2017.1-17.1%2017.1H-224c-.1.1-.1-34.2-.1-34.2z%22/%3E%3Cpath%20d=%22M-187.911%20419.9H-162v2h-25.911z%22/%3E%3C/svg%3E"
```

I would be thankful if someone could tell me how to replace the current thumbnail.

juanmard, 2020-07-18:
Very easy, open it and change the project properties... Link to example:

[Changed thumbnail](https://github.com/juanmard/gallery/blob/master/icestudio/changed_thumbails.gif)

AbsorbedInThought, 2020-07-18:
Oh okay, thank you!

closed 2020-07-18 by AbsorbedInThought