FPGAwars/icestudio #426: icestudio on a Raspberry Pi

rockyjsquirrel, 2020-07-28:
How did you run icestudio on a raspberry pi?
I've been trying for a while. hangs up in npm install, wrong version and such, then lots of errors which end up saying there is nothing wrong with npm. :(

Any chance you can release an SD Card image of that?

BTW, Awesome app for FPGAs !!!

cavearr, 2020-07-29:
Hi!! for the moment icestudio not works in raspberry pi but we are working on it  : )

rockyjsquirrel, 2020-08-05:
I know you closed the issue but this yuotube video is where I got my information.
   
  
  
   
  
  
   https://www.youtube.com/watch?v=ATtrY0CNmGw
  
  
   
  
  
   Thanks anyway for repsonding.

cavearr, 2020-08-05:
Yes this video is an old and primal version of icestudio, but now  it is not possible to run icestudio on raspberry pi, but very soon will be work, we are working in this way.

I'm maintaining this issue opened !

flegmatik, 2021-02-16:
Any news on this?

cavearr, 2022-04-26:
Hi! now you could download icestudio for linux arm64! https://downloads.icestudio.io

closed 2022-04-26 by cavearr