FPGAwars/icestudio #427: Toolchain instalation problem in Windows

Chenel000, 2020-07-30:
I am not possible to install ToolChain in my businees computer with Windows 10 Enterprise.

When I select Install ToolChain in the menu, when arrives 10% of the process I get in the low right side of the creen the error:
Command failed: C:\WINDOWS\system32\cmd.exe /s /c "python.exe ...

As I have different version of Python due to my work requirements, the operating system opens the first occurrence of Python in the Path, and it ssems to pen the Python 2.7 version. I have tried the same installation in my personal computer and if I set Python 3.5.8 to be the first in the Path, the ToolChain installation works correctly, so I assume this problem is due to the Python version.

I cannot modify the Path in my businees computer due to IT company restrictions, and I have commented previously I cannot remove other installations of Python.

I have tried to launch IceStudio from a cmd window with the addecuate Path variable, but this does not work because it seems this solution provides the right path to IceStudio, but IceStudio calls Python from another cmd window that uses the default Windows Path.

I see this a problem in order to install the ToolChain, I would suggest to be able to stablish the Python directory in a Preferences setting inside IceStudio in order to avoid this problem.

igor-m, 2020-09-10:
The same problem here with Win10. 
Python 2.7 and 3.7 and 3.8 installed. 
Quick hack:
1. go to Environment Variables -> System variables -> Path -> Edit Environment variable
2. move the path with Python 3.8 up, thus it is found before the 2.7 one.

cavearr, 2020-09-11:
Thanks! yes icestudio has a buggy python detection, this could be a possible solution. 

Enviados: Jueves, 10 de Septiembre 2020 21:09:03

qpeters, 2021-07-13:
@Chenel000  is this fixed in the latest stable release v0.6.0? https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0

Obijuan, 2021-07-15:
It would be great if you could test the 0.6.0 icestudio version ([Download it from here](https://github.com/FPGAwars/icestudio/releases/tag/v0.6.0)). Not sure if your problem is fixed, but if you test it and provide feedback we can solve it

Obijuan, 2021-07-15:
In addition, the toolchains can also be installed from the command line (outside icestudio). You will find the instruction in the documentation wiki: [Installing the Toolchains from outside icestudio](https://github.com/FPGAwars/icestudio/wiki/Installing-the-Toolchains-from-outside-icestudio)

cavearr, 2024-03-18:
I'm closing the issue, this is and old issue and is not aligned today.

If you need help about it , reopen it please!