FPGAwars/icestudio #428: Simple but infuriating problem?

FlyByPC, 2020-08-01:
Hi, all. Noob question here.

I want to use a clock input on more than one module. How can I connect one input pin to two output pins in Icestudio? This would be easy in MultiSim -- just wire from the second output pin to the input pin.

There's only one pin driving the net, so this should be legal. One clock pin drives two modules needing clocks.

I've spent about an hour on this, trying everything in the program I can think of, searching online, and I've finally wound up here.

It can't be this hard to connect one signal source to two sinks -- right??

Thanks.

ntegan, 2020-08-30:
That's pretty weird that you can't do this. 
I guess the only way would be to use the input/output labels.

https://streamable.com/0ns8ko

Obijuan, 2020-08-30:
You can do it with wires, very easily. Just place the second wire in a similar manner than the first, staring from the initial pin:

[image]

From the initial pin you can place all the wire you want to different pins

Obijuan, 2020-08-30:
You can also change the shape of the wire. Just place the cursor over the wire and drag it.  If you click on the x symbol you remove it:

In this gif you can see how the wire is reshaped, then removed, then placed again and given another shape

[image]

FlyByPC, 2020-08-30:
Thanks. It's working now, but I swear I tried that before and it simply would not add another wire. Can't reproduce the problem now, so I guess it's solved? Maybe a shift or control key was stuck or something, or the program got itself into a weird state somehow (although I did try restarting IceStudio.)

Anyway, thanks!

closed 2020-08-30 by FlyByPC

John-Westlake, 2020-12-22:
I have  exactly the same issue (Win8 0.5n201120) , it seems you have to place the cursor physically next to the device package output pin (before the orange junction "Dot" - touching the device body) to start placing the second track / junction. - this action for the second node is for sure different to placing the first track. Also pulling the tracks away from each causes havoc with many nodes being created and what a appears as a random rats nest of tracks doubling back on its self many times... AI auto-routing gone nuts...

When you watch the above video its all so well behaved, stable - behaves as you would hope... but for sure on my Win8 system it does not work as well as the in the video clip where apparently you can start the second junction node anywhere near the output device pin - so there is for sure something odd going on...

daddyzaur, 2021-04-29:
It is kind of difficult to add a wire or a junction in a schematic page. I have to assume there is a strong need to keep a tight integration between a schematic and the physical implementation of auto-routing inside the FPGA. Maybe the intent was to eliminate any chance of short-circuiting multiple outputs on a single electrical net? In any case, there are some hidden rules to understand:
    • Any Block has strong rules for automatic pin placement: Left Side for inputs, Right Side for outputs, and Top Side for controls
    • A wire can start only if the tip of the mouse cursor is placed over the physical (unique) output pin of a Block
    • A wire cannot be started from a Block input, or from the middle of a wire or a junction, must start from the physical (unique) output pin
    • Output pins and wires have same color, hence it is impossible to place the mouse tip over a pin if you do not zoom-in all the way to clearly see the pin
    • If a junction is needed, you must start a second wire from the same unique output, then slide the automatic junction where needed
    • If a wire connection is too convoluted, then it is possible to place wire labels where needed, and skip the curly wire

Patronics, 2024-08-21:
I found this confusing too, it could definitely benefit from a changing cursor, or some other clear indication that you must start to the left of the orange dot, rather than on it.