FPGAwars/icestudio #430: always @(posedge Up or posedge Dn)

Gerd56, 2020-09-03:
This statement:

always @(posedge Up or posedge Dn)

will verify, but not build. Building results in an error message.
To my limited knowledge this statement is correct. Is there a way around this problem?

Thanks

Obijuan, 2020-09-03:
That statement only makes sense in simulation, but not for real hardware. Any flip-flop only have one clock input, not two. There is no real hardware that matches that statement. For that reason the synthesizer tool complaints  
At the moment, Icestudio only works with real hardware and not in simulation

Gerd56, 2020-09-04:
I tried to build an up and down counter on my Nandland Go Board, not in a simulation.
This code verifies, but does not built:

reg [3:0] value;
always @(posedge Up or posedge Dn)
    if(Up)      value <= value + 1;
    else if(Dn) value <= value - 1;

This alternative code verifies and builds if used alone, but does not build together with a display block:

reg [3:0] value;
always @(posedge Up)
    value <= value + 1;
always @(posedge Dn)
    value <= value - 1;

Attached is the suspect Ice file. It had to be renamed .txt:

[build_fault2.txt](https://github.com/FPGAwars/icestudio/files/5172182/build_fault2.txt)

Please comment

Obijuan, 2020-09-04:
This second version has another problem: you are giving values to the same wire (value) from two different circuits. It is a shortcut. For that reason the building tool complaints

In correct digital designs, the first rule states that you should only use one clk signal, not two. Your up/down counter should have only one clk signal (the system clock) and two control signals: Up and Dn. A proper implementation would be:

reg [3:0] value = 0;

always @(posedge clk)
  if (Up)
    value <= value + 1;
  else if (Dn)
    value <= value - 1;

The Up and Dn control signals should last only one system clock period

This is a working example for the Go Board:

[image]

You can download the icestudio project from here:

[go-board-up-down-counter.zip](https://github.com/FPGAwars/icestudio/files/5173429/go-board-up-down-counter.zip)

When you upload it to the go-board, you will see the 4-bits value on the 4 leds (D4 is the least segnificant bit)  
When pressing the SW1 button, the counter will increase and you will see the new value on the leds. The same for the sw2 button, but in this case the counter is decreased by one

You can find more verilog and icestudio examples in the [Nandland repo](https://github.com/nandland/nandland)

Obijuan, 2020-09-04:
When working with Hardware description Languages, like Verilog, it is important to note that only a reduce sub-part of the language can be used for synthesize real hardware. Not all the statements are valid for describing real hardware. Many of them works perfectly on simulation, but not in hardware because there is no real hardware that matches that description. It takes some time to learn how to use Verilog for describing real hardware. The first step is to learn to think "in hardware". Then, once you have the real circuit in your mind, you describe it using verilog

Gerd56, 2020-09-04:
Good, I understand it now.
Thanks for your help.

closed 2020-09-04 by Obijuan