FPGAwars/icestudio #431: Program crashes on windows when using "Add as block"

androng, 2020-09-04:
When adding the file "counter.ice" to "counter-to-serial.ice" using "File > Add as block...", IceStudio 0.5.0 crashes. 
Windows 10 Pro 64-bit v1909 
[Counter-to-serial.zip](https://github.com/FPGAwars/icestudio/files/5176453/Counter-to-serial.zip)

androng, 2020-09-04:
Side note: program also crashes when copying and pasting a custom block such as "Counter" or "32-bit parallel-to-serial"

cavearr, 2020-09-04:
Try with the nightly version (really is more stable than "stable" version): 

[ https://github.com/cavearr/icestudio-nightly/raw/master/0.5.1n200523/icestudio-0.5.1n200523-win64.exe | https://github.com/cavearr/icestudio-nightly/raw/master/0.5.1n200523/icestudio-0.5.1n200523-win64.exe ] 

Enviados: Viernes, 4 de Septiembre 2020 20:48:22 
Asunto: [FPGAwars/icestudio] Program crashes when using "Add as block" (#431)

cavearr, 2021-08-09:
This should be fixed in the last wip, close the issue , if you detect in the last wips reopen please.

closed 2021-08-09 by cavearr