FPGAwars/icestudio #436: ULX3S progammer

goran-mahovlic, 2020-10-09:
Best is to use fujprog, as ujprog is stopped in time

https://github.com/kost/fujprog/releases

cavearr, 2020-10-09:
Yes! now that we have our new ulx3s, we can better integrate this board.

In the development branch with the latest development version of Apio, Icestudio uses fujprog instead of ujprog.

We will publish the new binaries soon!

Thank you!!!

goran-mahovlic, 2020-10-09:
@cavearr Thank you! Yes it took a while to get to you :) but it is finally there :1st_place_medal:

ladecadence, 2020-10-09:
In the last nightly build, icestudio tries to use fujprog, but it's not included in the binary tools of the ecp5 toolchain (just ujprog is there). So I needed to install fujprog myself and then it worked.

Obijuan, 2020-10-09:
You have to update the toolchain to version 0.5.5. Just click on
tools/toolchain/update toolchain. It will install the latest toolchain with
fujprog. Currently it only works for Linux but it will be implemented for
the other platforms soon

ladecadence, 2020-10-09:
Yeah, I tried that before, but it's not there.

`
% ls .icestudio/apio/packages/toolchain-ecp5/bin                         
bit_to_svf.py  ecppack  ecppll  ecpunpack  nextpnr-ecp5  ujprog  xsvftool-ft232h
`
This is on 0.5.1n200523

Obijuan, 2020-10-10:
You need to use the icestudio from the develop branch (clone the repo and
start it with npm install and npm start). The executable has not been
created yet, but it will be ready in a couple of days

In the mean time you have to start it with npm

cavearr, 2020-10-15:
@goran-mahovlic jajaja, thanks  for your work and support!!

Obijuan, 2020-12-03:
The fujprog is already included in the latest apio 6.0 and icestudio 0.5.1n201120. I am closing this issue

closed 2020-12-03 by Obijuan