FPGAwars/icestudio #440: ULX3S buttons

InverseInductor, 2020-10-17:
Currently only buttons B0, B3 and B6 work using 0.5.1n201009 on the ulx3s-85f.
Test was done using a simple connection between each button and each LED in order.
Button B0 has inverted logic, while B3 and B6 work as normal.
All other buttons only show a high logic signal.

I've confirmed that all LEDs work as intended.
I'm able to do get all buttons to work using standard symbiflow tools and a makefile.

Feels like a bad board constraints file, but I'm not sure where those are kept in icestudio.
B0 being inverted is explained in the [official constraints file](https://github.com/emard/ulx3s/blob/master/doc/constraints/ulx3s_v20.lpf), but it'd be nice to invert it. I can't find a good guide on lattice lpf files to adjust it myself just yet.

Obijuan, 2020-11-22:
Bug confirmed! It also happens with the ulx3s-12f (and possible with ulx3s-45f)  
I will get more information about it and let you know. Thanks for reporting the bug

Obijuan, 2020-11-22:
There is a bug on the generated .lpf file. Icestudio generates on the fly the .lpf file. The PULLMODE attribute is always set to NONE. This is a bug, as this attribute has different values for the input pints in the ulx3s board

Obijuan, 2020-11-23:
I've fixed it for the isolated pins. Now if you connect all the buttons directly to the leds (one by one, not using an array of buttons) it should work fine

[image]

I am attaching the icestudio file for testing

[test-ulx3s-12-btns.zip](https://github.com/FPGAwars/icestudio/files/5584888/test-ulx3s-12-btns.zip)

It would be great if you could test it. You need to test it from the development version

Obijuan, 2020-11-23:
I've also fixed it for array of pins. This example also works ok now:

[image]

This is the icestudio project file (for the ulx3s-12F)

[test-ulx3s-12-btns-array.zip](https://github.com/FPGAwars/icestudio/files/5585012/test-ulx3s-12-btns-array.zip)

Obijuan, 2022-01-17:
This issue is fixed

closed 2022-01-17 by Obijuan